UtilitechAS/amsreader-firmware #338: New sensor for estimated hourly consumption

eireit, 2022-10-23:
Is it possible to add a new sensor for estimated hourly consumption based on this formula:
Estimated=actual+derivert*(3600-seconds in the hour)/3600

More info can be found at this link: [](https://www.hjemmeautomasjon.no/forums/topic/9394-mitt-fors%C3%B8k-p%C3%A5-prediktiv-reduksjon-av-str%C3%B8mbruk/#comments)

ArnieO, 2022-10-23:
Not sure if I understand your formula (and quick glance at the link did not help; I might need to dig deeper). 

Is it the energy consumption [kWh] for the end of the ongoing hour-interval you want to estimate? 
- How do you calculate "derivert" at a certain point of time; some kind of continuously updated estimate?
- If so, the approach would only work under the assumption that the consumption for passed part of the hour-interval is representative for the remaining part of the hour-interval. What if a big load turns on (thermostat on water boiler, heated floor etc)?

gskjold, 2022-10-23:
I am reluctant to implement stuff like this because:
- It opens the door for all sorts of predictions that could be implemented (using precious memory)
- It also opens for endless discussions on validity of the calculation behind the prediction

This kind of data belongs in the home automation realm in my opinion.

dimnek, 2022-10-23:
If this is Home Assistant you could get a similar approximation there by defining one Riemann sum on a power sensor and wrap that with a utility meter helper:

Goto Settings -> Devices & Services -> Helpers -> Create Helper

Add Riemann sum integral sensor: Set input device to your power sensor. If your power sensor is in W and you want the output in kWh then select kilo as prefix and hour as time unit.

Add Utility Meter sensor: Use the Riemann sensor as input. Set meter reset cycle to 1 hour to get the value reset each hour.

https://www.home-assistant.io/integrations/integration/
https://www.home-assistant.io/integrations/utility_meter/

ArnieO, 2022-10-23:
> This kind of data belongs in the home automation realm in my opinion.

Agree.

gskjold, 2022-11-01:
Will not implement, belongs in home automation

closed 2022-11-01 by gskjold