UtilitechAS/amsreader-firmware #339: Data readout from encrypted Kamstrup meter fails with image > 1.5.8

jensen-2022, 2022-10-24:
I have tried setting up both an esp8266 and an esp32 using different AMS images:

Both works perfectly with the v1.5.8 image, but fails to read data using a newer image.

The esp8266 has been tried using 2.0.0, 2.0.10 and 2.0.11, the esp32 only 2.0.11 - all fail.

It seem like the data header for some reason is not recognized when using the newer AMS images.

**Hardware information:**
Denmark
Kamstrup
Encrypted
AMS reader generic ESP32 and generic ESP8266

**Relevant firmware information:**
 - Version: 2.0.10/2.0.11
 - MQTT: no
 - MQTT: NA
 - HAN GPIO: GPIO16 (esp32) GPIO13 (esp8266)
 - HAN baud and parity: 2400 8E1
 - Temperature sensors NA
 - ENTSO-E API enabled:  No

**Additional context**

When using the 1.5.8 image, the debug readout is:

(I) (HanReader)Got valid DLMS data (480 bytes)
(I) (HanReader)Full DLMS frame (489 bytes)
(I) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D 
(I) 45 01 CC 19  D9 82 01 D0  30 00 02 95  E8 B4 7C BB 
(I) D2 6F 91 BC  4F DA 71 2D  89 8D 1C 8F  D6 37 E8 C1 
(I) 64 22 A8 7A  7D 1B 49 6A  25 EE 97 59  A9 95 05 16 
(I) 93 25 95 1F  87 50 26 C8  D9 64 E7 63  07 0C C4 9D 

while the newer images return:

(W) (errorBlink) No HAN data received last 30s, single blink
(W) 7E 68 2F 0D  13 DC 32 E7  DB 68 05 4D  51 CC 46 27 
(W) 01 07 02 02  31 43 BC 4F  CA 3F 57 8D  74 A6 4B A2 
(W) 8D 24 85 61  83 6F D6 03  EE 4B C9 EB  5B 5A 2E D1 
(W) 03 E5 AD 2C  09 AD 52 78  90 C9 AA EA  A3 35 9A 23 
(W) A3 BB 51 33  A5 D6 A7 55  16 EB 0F 07  02 08 79 22 
(W) 98 C6 12 4E  3E 2E D2 38  3C 8F 7D 77  75 E9 49 8F 

Another example:

(W) (errorBlink)(C1) No HAN data received last 30s, single blink
(D) (AmsWebServer)Serving /data.json over http...
(V) (unwrapData)(C1) HDLC frame:
(V) 7E 68 2F 
(V) 7E 68 2F 0D  13 DC 32 E7  DB 68 05 4D  51 CC 46 27 
(V) 01 07 02 02  65 29 C9 BD  27 B8 50 52  09 EF 14 6A 
(V) BA C5 4E 37  66 DB BD BA  77 57 21 0D  C5 77 5A 53

gskjold, 2022-10-25:
Kamstrup is 8N1

jensen-2022, 2022-10-25:
Your absolutely right.

After setting the communication protocol to the correct settings, everything works perfectly.

I might suggest an addition to the FAQ section:

Q: My meter delivers data, but AMS reports No Data Received

A: Please check the communication parameters. If the interface is configured wrong, the data received will not be correct.

closed 2022-10-25 by jensen-2022