UtilitechAS/amsreader-firmware #341: Occational reboots

gskjold, 2022-10-25:
Some users report that they see reboots with 2.1.11, looking for input on how often. Please include information:
 - Country: [e.g. Norway]
 - Meter: [e.g. Aidon]
 - Encryption enabled [yes/no]
 - Hardware: [e.g. Pow-U, ESP32 etc]
 - MQTT: [yes/no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

revellion, 2022-10-26:
My Pow-K+ seems to have started to reboot continously, the only MQTT payload i'm able to get some times is the wifi RSSI before it spins around again. I had to go out this morning and remove it and re-seat it for it to kick up for a short while before going offline again. Wierdly enough it seems to be still connected to the unifi wifi since it's seen as a client there. but i can't reach the WebUI

Country: Sweden
Meter: Kamstrup DK-8660 Three-Phase meter
Encryption enabled: no
Hardware: Pow-K+ ESP32S2
MQTT: yes
MQTT payload type: Home-Assistant
HAN GPIO: [e.g. GPIO5]
HAN baud and parity: 2400 8N1
Temperature sensors none
ENTSO-E API enabled: no

ArnieO, 2022-10-26:
@revellion -> What RSSI do you read when you manage to connect? 
I know many meters in Sweden is located outdoors, could this be linked to bad Wifi reception?
Is this a new issue (firmware versions later than v2.1.8?
Potentially linked to https://github.com/gskjold/AmsToMqttBridge/issues/343#issuecomment-1291718636

revellion, 2022-10-26:
> @revellion -> What RSSI do you read when you manage to connect? I know many meters in Sweden is located outdoors, could this be linked to bad Wifi reception? Is this a new issue (firmware versions later than v2.1.8? Potentially linked to [#343 (comment)](https://github.com/gskjold/AmsToMqttBridge/issues/343#issuecomment-1291718636)

-91/-93 at it's lowest. -87 dbm at it's best. 

But i've gotten an acrylicplastic glasspanel waiting a cutout to replace the iron lid on the meterbox. and hoping it will be less RF unfriendly.

ArnieO, 2022-10-26:
Ok, that is indeed very low. Around -80 dBm seems to work well. I suspect wifi signal strength is the cause of your issues. Is it also a concrete wall in the direction towards your wifi router?

revellion, 2022-10-26:
> Ok, that is indeed very low. Around -80 dBm seems to work well. I suspect wifi signal strength is the cause of your issues. Is it also a concrete wall in the direction towards your wifi router?

Nah, it's connected to a Unifi UAP AC Outdoor Mesh AP on the other side of a wooden garage. so not too much in the way RF wise. The worse is probably the metalbox the meter is installed on the pole of the incoming service.

ArnieO, 2022-10-26:
> Nah, it's connected to a Unifi UAP AC Outdoor Mesh AP on the other side of a wooden garage. so not too much in the way RF wise. The worse is probably the metalbox the meter is installed on the pole [...]

OK, agree - probably the cabinet then. Indoor cabinets are surprisingly transparent to 2,4 GHz, probably because of clearings at the cable feedthroughs.

revellion, 2022-10-26:
-71dbm with acryllic door replacement, and a green "wifi" text in the WebUI seems like a sane indicator 🗡️

ArnieO, 2022-10-26:
> -71dbm with acryllic door replacement, and a green "wifi" text in the WebUI seems like a sane indicator 🗡️

That's more like it! 👍

gskjold, 2022-10-27:
Please consider testing firmware in this post: https://github.com/gskjold/AmsToMqttBridge/issues/343#issuecomment-1293843402 as it could relate to this issue. Install, go to wifi config and check that power saving is set to default. Test that for a while and if it works I would also recommend trying the "Min" setting. Currently the firmware defaults to "Max" starting from v2.1.8.

gskjold, 2022-11-01:
How is v2.1.12 treating you?

revellion, 2022-11-06:
Much more stable here, uptime is at tops 1 day. But it's generally more responsive.

Though i've seem to notice if my MQTT broker is offline for a bit too long that the Pow-K+ locks up and gets unresponsive. dunno if it queues up too much in memory and locks up or something