UtilitechAS/amsreader-firmware #342: Meter configuration: Manufacturer, Model, ID not selectable

sirtet, 2022-10-25:
Beginner Question:
After flashing and WIFI setup i can connect to the Web UI.
Going to Configuration > Meter, i can adjust Baudrate etc, but not the Meter Manufacturer, Model, ID.
I have not yet connected the Meter (KAMSTRUP Omnipower), as i did not get the Keys yet, and i suspect push is also not yet enabled.

Are these settings only available after some serial Data is received, or should i already be able to select something? After reading the WIKI, i would think so.

**Hardware information:**
 - Country: Switzerland
 - Meter: Kamstrup
 - Encryption enabled: yes
 - AMS reader: ESP32 
 - M-bus adapter (if applicable): NO

**Relevant firmware information:**
 - Version: 2.1.1
 - HAN GPIO: GPIO16
 - HAN baud and parity:  2400 8E1

gskjold, 2022-10-25:
Thank you for pointing this out, the wiki have not been updated for the later updates. Manufacturer, model and ID are autodetected and not selectable. 2400 8N1 should be sufficient configuration for you

gskjold, 2022-10-26:
Updated wiki

closed 2022-10-26 by gskjold