UtilitechAS/amsreader-firmware #343: Permanently slow Pow-U after long-lasting (and ongoing) mqtt server outage (+ entsoe outage)

oivindoh, 2022-10-25:
Running fw 2.1.10 on a new style (esp32-s2) pow-u connected to a kaifa meter, the ams reader startet responding slowly after both failing to deliver mqtt data (due to missing server) and not being able to fetch entsoe data after their outage.

Using the web interface, operations either time out or respond very slowly, and sections can display "Data table is not defined." repeatedly instead of actual data.

I can't even get debugging enabled because the requests to do so keep timing out.

I did manage to upgrade the device to 2.1.11, but the slow issue remains, even after a factory reset performed after said upgrade.

Here's a screenshot showing the situation (after factory restore and only connected to power via usb):

[image]

**To Reproduce**

I'm unsure whether entsoe outage or mqtt outage is the culprit here, but I'm kind of assuming this is due to the device being overloaded by queued mqtt data, even though mqtt was disabled (and still is) a while after the outage was discovered.

**Hardware information:**

 - Country: Norway
 - Meter: Kaifa
 - Encryption enabled no
 - AMS reader: Pow-U esp32-s2

**Relevant firmware information:**

 - Version: 2.1.10, 2.1.11
 - MQTT: yes
 - MQTT payload type: json
 - HAN GPIO:
 - HAN baud and parity: 2400 8E1
 - ENTSO-E API enabled: yes

oivindoh, 2022-10-25:
I deployed a new "emergency" mosquitto server and after a few failed/stalled requests to do so, the device connected to mqtt and appeared to resume normal operation.

Here's a snip from a standard ping I had running against the unit when things were'nt looking so rosy (but I could still get a request across after several retries):

```
64 bytes from 192.168.1.2: icmp_seq=503 ttl=255 time=4358.124 ms
64 bytes from 192.168.1.2: icmp_seq=504 ttl=255 time=3354.566 ms
Request timeout for icmp_seq 509
Request timeout for icmp_seq 510
Request timeout for icmp_seq 511
Request timeout for icmp_seq 512
64 bytes from 192.168.1.2: icmp_seq=508 ttl=255 time=5346.242 ms
64 bytes from 192.168.1.2: icmp_seq=509 ttl=255 time=4341.031 ms
64 bytes from 192.168.1.2: icmp_seq=514 ttl=255 time=628.225 ms
Request timeout for icmp_seq 516
Request timeout for icmp_seq 517
```

Now there are no lost ones, though latency is still in the hundreds of ms to 1-2 seconds range. I suppose this is normal for this kind of unit - I don't have a plot of that from before the outage started.

oivindoh, 2022-10-25:
So, back to the insecurity around cause in the issue description, I guess I'm concluding this is caused by a long-lasting mqtt outage

tha1971, 2022-10-26:
After upgrading from 2.1.9 to 2.1.11 I lost Entsoe.
But then in config Entsoe I switched currency from dkk to euro ( that worked) and back to dkk then it worked again with dkk.
Now it’s working as normal.
Best regards Thomas

gskjold, 2022-10-26:
Interesting, I had a similar experience with my esp32-s2 a few weeks ago that I was not able to identify the cause of. I have been waiting for the problem to return so that I could debug it, but has not happened since.

ArnieO, 2022-10-26:
Interesting indeed. I too believe I have noticed a somewhat slower response from versions after 2.1.8. But difficult to know whether it is caused by my Wifi network (TP-link mesh system; quite variable performance). 
This is "typical" now:
[image]

@gskjold : Could there be a link here with the "level" of the Autosleep-setting that was introduced recently? If so - the ESP32-S2 Pow-U needs the current setting when powered from HAN/M-bus. But other devices (Pow-K, Pow-P1, homemade units that are not powered from M-bus) could then potentially benefit from a less "severe" auto sleep setting?

gskjold, 2022-10-26:
I had the exact thought, maybe make that user configurable, use max for default on BUS powered while min on the rest.

ArnieO, 2022-10-26:
> I had the exact thought, maybe make that user configurable, use max for default on BUS powered while min on the rest.

Yes, that sounds like a good solution. And it means one more parameter we need to document... 😅

mikkle, 2022-10-26:
Regarding wifi sleep levels:
The explicit setting of modem_sleep_min/max has completely broken stability on my Pow-K+ (DK, Kamstrup, Encr, /Entso-e, mqtt (raw full))
The last stable release was 2.1.3 - before the set_sleep commands were introduced in wifi_connect().
I'm currently running a customized build (2.1.11 + removed line 1057 in AmsToMqttBridge.ino (		WiFi.setSleep(WIFI_PS_MAX_MODEM);)
This gives me a pretty stable experience. Without this patch, I get reboots approx every 15 minutes, and the extremeliy high latency reported above).
I've been trying to get my head around the wifi sleep documentation from Espressif, but it seems rather opaque and contradictory. The only conclusion I can glean with 100% certainty is that WIFI_PS_MIN_MODEM is default on esp32 (and -s2).

Some parts of the dox seem to suggest that setSleep should be called _after_ connecting to the AP, but this is one of the points that seem to be contradicted in the same doc. (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html)
I have tried a couple of custom builds where I set PS_MIN_MODEM/PS_MAX_MODEM after wifi.init(), but that had no effect.
As soon as the setSleep method is introduced - no matter if min/max, or before/after wifi_init(), the result is high latency + constant reboots af the mentioned 15-ish minutes.
Maybe it is a framework problem (like with the missing ap_mode in esp32-s2)? - After all, this code uses the arduino-frameworks and does not call the low-level esp wifi api directly...

Sorry for the messy comment :)
Please let me know If I can contributed with some more structured tests, or if you need experimental builds given a spin :)

Cheers!
Mikkel

ArnieO, 2022-10-26:
@mikkle Thank you for great input! 👍 (and not at all messy)

@gskjold and myself have tested code with `WiFi.setSleep(WIFI_PS_MAX_MODEM);` before introducing it in the "production" code. We know it has a measurable and significant effect on power consumption, and is needed on devices that are powered from M-bus (notably the Pow-U) - which is the reason for implementing it in the code. Without it, the Pow-U with ESP32-S2 is not able to keep up its voltage. 

I have been running firmware with `WiFi.setSleep(WIFI_PS_MAX_MODEM);` on my Pow-K+ for a while now without significant negative indication, except the mentioned slow response (sometimes), which I have thought could be due to my wifi. I do not see the reboot problem you mention.

Since introducing `WiFi.setSleep(WIFI_PS_MAX_MODEM);` I have shipped around 200 Pow-U and Pow-K+ devices, and there are only few feedbacks like yours. This indicates that what you experience is not a common issue. But your issues and the testing you have done seems indeed to reveal a vulnerability in the code - maybe in certain wifi environments.

As you say, existing documentation is vague - so this is a tricky point.

Let us wait for the thoughts of @gskjold, but I my thinking is: Until we learn more about the effects of this code line, the best option is probably to make it user configurable, so that HW that does not need `WiFi.setSleep(WIFI_PS_MAX_MODEM);` can switch it off.

mikkle, 2022-10-26:
Right,

The high latencies seem to make sense because if how PS_MAX_MODEM works (wake up on fixed intervals in stead of on DTIM beacons from AP). In other words, the ECHO REQ packets hang around in the ESP wifi input buffer for a random-ish time (max being until next fixed wakeup). PS_MIN_MODEM has much more frequent wakeups, hence the generally later latency.

I know for certain that my wifi environment is quite noisy - both in terms of channel-noise and in terms of broadcasts, so that might indeed also be a factor.
I'll try to re-work my wifi-setup so the pow-k gets its own broadcast domain (but that means internet downtime for the kids, so I'll have to obtain a maintenance window :-)

Meanwhile, I'm now testing a build, where I again have moved WiFi.setSleep(WIFI_PS_MAX_MODEM); to right after WiFi.begin() - before the yield() in that block (lines 1119-1120-ish) because I don't really recall if I had the setSleep before or after the yield() in my earlier tests.

This approach seems to match Espressif's own sample code:
https://github.com/espressif/esp-idf/blob/009d15e59db4f50fcf08849f14427c8ebbc1aff4/examples/wifi/power_save/main/power_save.c

They have the esp_wifi_set_ps() immediately after esp_wifi_start(), and as I read thier sample, wifi_start() triggers the event_handler, where wifi_connect() is called.

The arduino lib calls wifi_connect() at the end of wifi.start(), so this would seem to be the proper way of doing it.

I'll report back with uptime findings - might take some time for the first crash (I seem to remember a handful of hours, then repeatedly the above-mentioned 15ish minutes afterwards)
(actually, here is a graf of how it looked on one occasion (can't remember the exact version I tested then, but something >2.1.3)
[image]

Handl3vogn, 2022-10-27:
Just want to add that I have similar problems on my pow-k+ (esp32-s2).

Really slow web ui and mqtt disconnecting at times. If i unplug and plug it back inn it seems to be stable for many 6 hours before ut becomes unstable again.

gskjold, 2022-10-27:
Thank you @mikkle for the valuable input. I will make a test version soon where sleep mode is optional and move wifi.setsleep after wifi.begin

mikkle, 2022-10-27:
Cheers @gskjold ! Haven't installed it yet (will test the various settings and report back), but the changes looks to be spot on!
Nice work, thanks! :-)

Handl3vogn, 2022-10-28:
Seems to have the same experience as @mikkle 
Changing from default to min made the web ui and ping respond much faster. 

And have been running for 12 hours now and so far no slowdowns or mqtt connection issues.

ArnieO, 2022-10-28:
My current consumption measurements confirm that MIN_MODEM is not default behavior on the S2.
Measured on a Pow-U with ESP32-S2:
[image]

oivindoh, 2022-10-29:
I just had a self-inflicted outage on my MQTT server gain, and noticed the web GUI was acting extremely sluggish.

Here's a ping session that shows the transition from MQTT server being unavailable to MQTT server being available:

```
ping 192.168.1.2
PING 192.168.1.2 (192.168.1.2): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4
64 bytes from 192.168.1.2: icmp_seq=0 ttl=255 time=5675.750 ms
64 bytes from 192.168.1.2: icmp_seq=1 ttl=255 time=4674.238 ms
<MQTT server is now available on the network>
64 bytes from 192.168.1.2: icmp_seq=2 ttl=255 time=3669.076 ms
64 bytes from 192.168.1.2: icmp_seq=3 ttl=255 time=2664.778 ms
64 bytes from 192.168.1.2: icmp_seq=4 ttl=255 time=1660.347 ms
64 bytes from 192.168.1.2: icmp_seq=5 ttl=255 time=657.455 ms
64 bytes from 192.168.1.2: icmp_seq=6 ttl=255 time=6.329 ms
64 bytes from 192.168.1.2: icmp_seq=7 ttl=255 time=5.983 ms
64 bytes from 192.168.1.2: icmp_seq=8 ttl=255 time=7.191 ms
64 bytes from 192.168.1.2: icmp_seq=9 ttl=255 time=573.707 ms
64 bytes from 192.168.1.2: icmp_seq=10 ttl=255 time=1014.797 ms
64 bytes from 192.168.1.2: icmp_seq=11 ttl=255 time=9.676 ms
64 bytes from 192.168.1.2: icmp_seq=12 ttl=255 time=950.788 ms
64 bytes from 192.168.1.2: icmp_seq=13 ttl=255 time=358.439 ms
64 bytes from 192.168.1.2: icmp_seq=14 ttl=255 time=1007.046 ms
64 bytes from 192.168.1.2: icmp_seq=15 ttl=255 time=7.076 ms
64 bytes from 192.168.1.2: icmp_seq=16 ttl=255 time=984.771 ms
64 bytes from 192.168.1.2: icmp_seq=17 ttl=255 time=7.261 ms
64 bytes from 192.168.1.2: icmp_seq=18 ttl=255 time=1475.699 ms
64 bytes from 192.168.1.2: icmp_seq=19 ttl=255 time=472.270 ms
64 bytes from 192.168.1.2: icmp_seq=20 ttl=255 time=1004.253 ms
64 bytes from 192.168.1.2: icmp_seq=21 ttl=255 time=6.885 ms
64 bytes from 192.168.1.2: icmp_seq=22 ttl=255 time=977.821 ms
64 bytes from 192.168.1.2: icmp_seq=23 ttl=255 time=4.786 ms
64 bytes from 192.168.1.2: icmp_seq=24 ttl=255 time=919.269 ms
64 bytes from 192.168.1.2: icmp_seq=25 ttl=255 time=5.736 ms
64 bytes from 192.168.1.2: icmp_seq=26 ttl=255 time=947.373 ms
```

mikkle, 2022-10-29:
@ArnieO 
After thinking a bit, my new working hypothesis is that MIN_MODEM being the default according to the documentation might simply mean that if setSleep() is called with just bool:true as the argument, MIN_MODEM is the behaviour that will be applied.
Sleep modes also have visible effect on measured VCC:
[image]
For the record, I calibrated vcc on the pow-k+ (setting offset to 2.7 - no multiplier) while on MAX_MODEM.

Another interesting point is that the Wireless environment really does seem to have an effect as well. Notice that Vcc rose, when the pow-k+ roamed to another AP. For reference, here are the recorded RSSI values for the same period:
[image]
Signal is a bit better after the roaming event, at that seemed to trigger a rise in measured VCC.

Cheers!

ArnieO, 2022-10-30:
@mikkle : Very interesting indeed - but also a bit difficult to quite understand what is going on here.

I have never had much focus on Vcc on my own Pow-K+ (still using a layout v1.5, which cannot read Vcc...), so this is new information for me. I'll obviously have to switch to a v1.51 Pow-K+ to keep an eye on Vcc.

Vcc calibration should in principle not be needed on the ESP32, as I believe the code uses factory-burnt parameters for calibration (right, @gskjold ?). So I stopped pre-calibrating during initial programming when going from ESP8266 (pre Pow-K+ designs) to ESP32. But I agree with your approach: To read Vcc as accurately as possible, you can measure Vcc with a good multimeter, then adjust offset to calibrate out production variations in the resistor divider that sits on the ADC input.

mikkle, 2022-10-30:
@ArnieO "default (no ps)" means I've set the new Wifi Power Saving config option to "default" meaning the code never calls setSleep()
The config between those two markers are still the same (Configured as "default"). The first of those markers (8:30-ish) indicates when the Pow-K+ roamed to a diffrent AP. At The later marker (11:00-ish) I changed the power saving config to "Min" (PS_MIN_MODEM getting set in the code).

Btw, it just says "V1.5" on my Pow-K+, so mayby my Vcc readings are totally mis-leading?
[image]
[image]

oivindoh, 2022-10-30:
I noticed the device got really unresponsive again today as I wanted to check on some weird price grahps I recall seeing for tonight.

The tail end of a long ping session towards the device

```
Sun Oct 30 16:11:51 CET 2022: Request timeout for icmp_seq 342
Sun Oct 30 16:11:52 CET 2022: Request timeout for icmp_seq 343
Sun Oct 30 16:11:53 CET 2022: Request timeout for icmp_seq 344
Sun Oct 30 16:11:54 CET 2022: Request timeout for icmp_seq 345
Sun Oct 30 16:11:55 CET 2022: Request timeout for icmp_seq 346
Sun Oct 30 16:11:56 CET 2022: Request timeout for icmp_seq 347
Sun Oct 30 16:11:57 CET 2022: Request timeout for icmp_seq 348
Sun Oct 30 16:11:58 CET 2022: Request timeout for icmp_seq 349
Sun Oct 30 16:11:59 CET 2022: Request timeout for icmp_seq 350
Sun Oct 30 16:12:00 CET 2022: Request timeout for icmp_seq 351
Sun Oct 30 16:12:01 CET 2022: Request timeout for icmp_seq 352
Sun Oct 30 16:12:02 CET 2022: Request timeout for icmp_seq 353
Sun Oct 30 16:12:03 CET 2022: Request timeout for icmp_seq 354
Sun Oct 30 16:12:04 CET 2022: Request timeout for icmp_seq 355
Sun Oct 30 16:12:05 CET 2022: Request timeout for icmp_seq 356
Sun Oct 30 16:12:06 CET 2022: Request timeout for icmp_seq 357
Sun Oct 30 16:12:07 CET 2022: Request timeout for icmp_seq 358
Sun Oct 30 16:12:07 CET 2022:
Sun Oct 30 16:12:07 CET 2022: --- 192.168.1.2 ping statistics ---
Sun Oct 30 16:12:07 CET 2022: 360 packets transmitted, 11 packets received, 96.9% packet loss
Sun Oct 30 16:12:07 CET 2022: round-trip min/avg/max/stddev = 9.009/43.311/175.168/45.844 ms
```

The wifi network is performing well otherwise, with no interference to speak of, and excellent response times for the other clients connected.

Subscribing to the configured mosquitto topic, I can see the device is happily pushing data to it while the device is almost entirely unavailable through web interface:

```
2022-10-30 16:11:40.659936: {'lv': 'KFM_001', 'id': '6970000000000095', 'type': 'MA304H3E', 'P': 1943, 'Q': 0, 'PO': 0, 'QO': 844, 'I1': 4.96, 'I2': 6.87, 'I3': 4.37, 'U1': 239.8, 'U2': 239.6, 'U3': 239.4}
2022-10-30 16:11:42.183438: {'P': 2051}
2022-10-30 16:11:44.243695: {'P': 2292}
2022-10-30 16:11:46.187523: {'P': 2007}
2022-10-30 16:11:48.265410: {'P': 1728}
2022-10-30 16:11:50.696658: {'lv': 'KFM_001', 'id': '6970000000000095', 'type': 'MA304H3E', 'P': 1967, 'Q': 0, 'PO': 0, 'QO': 801, 'I1': 5.12, 'I2': 7.21, 'I3': 3.93, 'U1': 239.6, 'U2': 239.45, 'U3': 239.3}
2022-10-30 16:11:52.181083: {'P': 1958}
2022-10-30 16:11:54.276911: {'P': 1758}
2022-10-30 16:11:56.149469: {'P': 2052}
2022-10-30 16:11:58.191053: {'P': 1779}
2022-10-30 16:12:00.730108: {'lv': 'KFM_001', 'id': '6970000000000095', 'type': 'MA304H3E', 'P': 1817, 'Q': 0, 'PO': 0, 'QO': 821, 'I1': 4.64, 'I2': 6.74, 'I3': 4.08, 'U1': 239.6, 'U2': 239.5, 'U3': 239.4}
2022-10-30 16:12:02.216605: {'P': 2095}
2022-10-30 16:12:04.164285: {'P': 1793}
2022-10-30 16:12:06.159228: {'P': 1969}
```

For completeness sake; a screenshot of the GUI interface managing to show up with at least some data after quite a few attempts:

[image]

ArnieO, 2022-10-30:
@mikkle 
> Btw, it just says "V1.5" on my Pow-K+, so maybe my Vcc readings are totally mis-leading?

You have the same as the one I'm currently using. 
It has a resistor divider on GPIO35:
[image]
This was an unfortunate mistake inherited from a design for ESP32. On the ESP32-S2, it is one of the IOs that cannot be configured as ADC:
[image]
On layout v1.5.1 it was moved to GPIO10:
[image]

So yes, I am surprised that you get "sensible" Vcc readings from your Pow-K+ layout v1.5. Which GPIO are you reading (System/GPIO->Vcc)?

oivindoh, 2022-10-30:
I've now reseated the device on a different AP (channel 3+1 vs channel 8), and latency/responsiveness is much improved, with GUI being fast, and ping averaging 37ms, stddev 16ms, even though RSSI dropped to ~ -80dBm.

I then updated to the firmware above above, which was still similar since I have yet to connect it to the meter or make a change in the sleep configuration.

I then made MQTT unavailable, which seems to have no effect on Pow-U availability. I guess my issue may have been the AP all along. That's what you get for unwittingly betatesting Unifis wifi6 products, I guess...

ArnieO, 2022-10-30:
@oivindoh Thank you! It is indeed difficult to see a pattern on what makes the devices more or less responsive. I have a gut feeling it is often "something" with the Wifi environment, but that is just my speculation.

mikkle, 2022-10-30:
@ArnieO @oivindoh Actually, I'm also using Ubiquiti/Unifi wireless APs, albeit earlier generations, so maybe that's a weird common denominator?
@ArnieO Pow-K+ configured to read Vcc from GPIO 10 (the sw defaulted to that in commit https://github.com/gskjold/AmsToMqttBridge/commit/cab6c54ed98e5c8a2a0828286d68ef2023ca29cf)
Is it just ambient stuff I'm picking up on GPIO 10 you think?

ArnieO, 2022-10-30:
> @ArnieO Pow-K+ configured to read Vcc from GPIO 10 (the sw defaulted to that in commit [cab6c54](https://github.com/gskjold/AmsToMqttBridge/commit/cab6c54ed98e5c8a2a0828286d68ef2023ca29cf))
> Is it just ambient stuff I'm picking up on GPIO 10 you think?

Yes, that cannot be valid data. It is your large offset value plus "something".
Here is what I get when setting Vcc to 10:
[image]
[image]

gskjold, 2022-11-01:
I'm assuming all of your issues have been fixed by 2.1.12 ?

oivindoh, 2022-11-01:
From my perspective this isssue is closed; not entirely sure about the other stuff going on here?