UtilitechAS/amsreader-firmware #345: Suggestions for UI improvement

sirtet, 2022-10-27:
This is not a feature-request per se, just me trying to be constructive, don't feel pressured...
I very much like the current UI, it's an awesome project, thanks a lot.
Still, here are my thoughts on how to improve even further. It's just my view and maybe should be ignored ;-)

The menu-bar and the status bar below it both show status info.
That could be consolidated into one status-area.
I would suggest to not show rssi etc. on the main page, as that's more debug-info vs. all the metering-info below.
Instead, create a new status-page that has a few sections relating to the current colored status indicators like HAN etc.
Make the current indicators clickable, linked to the sections in the status page.
Maybe have only one status indicator "Status: OK" "Status: 2 Issues".
This would create a cleaner looking UI with less distraction from the actual meter-data.

Also, why is __System__ a separate menu-entry? Looking at the Items in it, it seems to fit very well into __Configuration__.

gskjold, 2022-10-27:
Thank you for some good input here, much appreciated. I have a v2.2 in the works with a new (but looks mostly the same) UI where I have consolidated the status bar and menu bar. The reason why voltage, rssi, memory and time have been presented on the main page is mostly for me actually; When people report bugs, they tend to also add a screenshot, and this info sometimes help identify problems. I like the idea of a extended status page, I will keep that in mind

gskjold, 2022-11-21:
Mostly there in upcoming 2.2

closed 2022-11-21 by gskjold

gskjold, 2022-12-06:
v2.2 beta out for testing now:
https://github.com/gskjold/AmsToMqttBridge/discussions/370