UtilitechAS/amsreader-firmware #347: Aidon 6242 (Sweden)

toast1234, 2022-10-28:
Does this look like something that can be used?
Honestly I dont even know if it is the right protocol. But it is not DSMR

**Hardware information:**
 - Country: Sweden
 - Meter: Aidon
 - Encryption enabled dont know
 - AMS reader: ESP32
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.1.0]
 - MQTT: [yes/no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: GPIO18
 - HAN baud and parity: 115200 8N1
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

W) (errorBlink)(C1) No HAN data received last 30s, single blink
(E) (unwrapData)(C1) Ended up in default case while unwrapping...
(E) 80 E0 38 0A  F0 E1 FE FE  FF FE FD C9  FF FE FD FE 
(E) FE BD 20 0E  FE ED E2 F8  FE F8 F1 FE  3C FF F9 F2 
(E) F2 C0 FE 82  80 F2 F9 C8  F3 F8 FE F2  F1 FE BE FF 
(E) FE FE F2 F2  C0 FE 82 80  F2 F0 C8 F3  F8 FE F0 E0 
(E) 3C 33 90 F2  C0 92 F0 C8  E2 F8 FE 80  E0 3C FE F8 
(E) C0 F2 F2 E1  FE 82 C9 F2  F0 C8 E2 F8  FE 82 E0 FE 
(E) 3C FE FE FE  F2 F2 C0 FE  82 80 F2 FE  80 E0 FE 3C 
(E) FE FE FE F2  F2 C0 FE 82  80 F2 F0 C8  F3 F8 FE 8E 
(E) E0 FE 3C FE  FE 3E FE 82  80 F2 F0 C8  E2 F8 FF 08 
(E) E0 FE 3C FE  FE F2 FE 82  80 F2 F0 C8  E2 F8 FE 02 
(E) F1 FE 3C FE  FE FE F2 F2  C0 FF 82 80  F2 F0 C8 E2 
(E) F8 FE 00 E0  FE 3C FE FE  FE F2 F2 C0  FE 82 C5 F2 
(E) F0 C8 E2 F8  FE 06 E0 FE  BE FE FE F2  F2 C0 FE 82 
(E) 80 F2 F9 C8  E2 F8 FE 00  E0 FE 3C FE  FB 9E F2 F2 
(E) C0 FE 82 80  F2 F0 C8 F3  F8 FE 02 E0  FE 3C FE FE 
(E) FE FB F2 C0  FE 82 80 F2  F0 C8 E2 F8  FF 00 E0 FE 
(E) 3C FE FE FE  F2 F2 E1 FE  82 80 F2 F0  C8 E2 F8 FE 
(E) F1 FE 3C FE  F8 F2 F2 C0  FE D3 80 F2  F0 C8 E2 F8 
(E) FE F8 CE FE  3C 8E 12 3E  FE 82 82 FB  F0 C8 E2 F8 
(E) FE F2 CE FE  3C FE FE FE  F2 F2 C0 FE  82 82 F2 F0 
(E) ED E2 F8 FE  F0 CE FE 3C  F2 24 F2 C0  FE 82 3E F2 
(E) F0 C8 E2 FD  FE E6 CE FE  3C C2 0E FB  F2 C0 FE 82 
(E) 3E 30 
(W) (errorBlink)(C1) No HAN data received last 30s, single blink
(W) (errorBlink)(C1) No HAN data received last 30s, single blink
(E) (unwrapData)(C1) Ended up in default case while unwrapping...
(E) 80 F5 38 F0  F8 FF FE FF  FD 3E FF FF  F8 FE FF 38 
(E) 20 12 FC C8  E2 F8 FE F8  E0 FE 3C FE  F0 F2 F2 C0 
(E) FE 82 80 FB  F0 C8 E2 F8  FE F2 E0 FE  3C FE FE FE 
(E) F2 FB C0 FE  82 80 FB F0  ED E2 F8 FE  F0 E0 FF 3C 
(E) FF C8 E2 F8  FE 06 E0 FE  32 90 F2 C0  92 F0 C8 F3 
(E) F8 FE E0 FE  32 F2 E1 92  F0 C8 E2 F8  FE 80 F1 FE 
(E) 3C FE F8 C0  F2 F2 C0 FE  C8 E2 F8 FE  82 E0 FE 3C 
(E) FE FE FE F2  F2 C0 FE 82  80 FB F0 C8  E2 F8 FE 80 
(E) E0 FE 3C FE  FE FE F2 F2  C0 FE 82 80  F2 F0 ED FE 
(E) 8E E0 FE 3C  FE E0 FE FE  82 80 F2 F0  C8 E2 FD FE 
(E) 08 E0 FE 3C  FE FE FE 82  80 F2 F0 C8  E2 F8 FE AB 
(E) E0 FE 3C FE  FE FE F2 F2  C0 FE 82 80  F2 F0 C8 E2 
(E) F8 FE 00 E0  FF 3C FE FE  FE F2 F2 C0  FE 82 80 F2 
(E) F0 C8 E2 F8  FE 06 E0 FE  3C FE FE F2  F2 C0 FE 82 
(E) 80 FB F9 C8  E2 F8 FE 00  E0 FE 3C FE  F2 C0 F2 F2 
(E) C0 FE 82 80  F2 F0 C8 F3  F8 FE 02 E0  FE 3C FE FE 
(E) FF F2 F2 C0  FE 82 80 F2  F0 C8 E2 F8  FF 00 E0 FE 
(E) 3C FE FE FE  F2 FB C0 FE  82 80 F2 F0  C8 E2 F8 FE 
(E) F1 FE 3C FE  F8 F2 F2 C0  FF 82 80 F2  F0 C8 E2 F8 
(E) FE F8 CE FE  3C 8E 12 3E  FE 82 C3 F2  F0 C8 E2 F8 
(E) FE F2 CE FE  3C FE FE FE  F2 F2 C0 FE  82 82 F2 F9 
(E) C8 E2 F8 FE  F0 CE FE 3C  F2 24 F2 C0  FE 82 3E F2 
(E) F0 C8 F3 F8  FE E6 CE FE  3C C2 0E FB  F2 C0 FE 82

ArnieO, 2022-10-28:
@toast1234 
Are you aware that you need a pullup resistor on the P1 port data line, as it is a Common collector output? 3k to 3,3V or 4k7 to 5V will work. 

I have heard of one case where activating the internal pullup on the GPIO was sufficent, but I consider that "pure luck", as it is in the order or 25-30k.

gskjold, 2022-10-28:
This software supports both. Your debug output suggests there is data corruption.

toast1234, 2022-11-01:
Working after re-routed GPIO18 to GPIO16 and inverted signal (115200, 8N1, inverted)

closed 2022-11-01 by toast1234

ArnieO, 2022-11-01:
> Working after re-routed GPIO18 to GPIO16 and inverted signal (115200, 8N1, inverted)

Right - you need pullup AND inversion (either in hardware or in code).