UtilitechAS/amsreader-firmware #348: Document Home Assistant MQTT sensors

shadowano, 2022-10-30:
**Is your feature request related to a problem? Please describe.**
The Wiki is missing documentation of what data structure/sensors is sent when choosing Home Assistant from the MQTT settings.
I want more data then what is currently chosen for HA, so I'm trying to create sensors manually from the Raw values (full), but e.g. current power usage (import) shows a strange bar above the line graph.  Below is how I have configured it, and I guess I'm just missing something here.

[image]

`mqtt:
  sensor:
    - state_topic: "ams/reader/data/meter/import/active"
      name: "Current power usage"
      unique_id: "ams.reader.data.import.active"
      state_class: measurement`

**Describe the solution you'd like**
A clear documentation might help others if they want to manually add sensors correctly. At the same time, I'm missing the following HA data to be sent to keep it simple for auto discovery :) 
- {publish topic}/realtime/import/hour
- {publish topic}/realtime/import/day
- {publish topic}/realtime/import/threshold
- {publish topic}/realtime/import/monthmax

gskjold, 2022-10-31:
Unfortunately I have absolutely no experience with HA, so I need someone to pitch in on HA specifics.

As for documentation, I can add basic info about the sensors and the JSON data it is backed by.

You say you are missing realtime data from raw full mode? What kind of MQTT broker do you use? There was recently a similar issue, where the problem was that the broker had limitations in number of topics.

shadowano, 2022-10-31:
No, I'm missing real time data from HS payload. Btw, I'm using version 2.1.8, not sure if you have added more info in latest version:) 

My broker is mosquito, recommended by HS. I see all data when using full raw.

ArnieO, 2022-10-31:
> I'm using version 2.1.8, not sure if you have added more info in latest version:)

You should update (and read the release-comments :wink:) https://github.com/gskjold/AmsToMqttBridge/releases

shadowano, 2022-10-31:
Yes, i should @ArnieO, and I just now did :) But I also have the impression that newer versions are not that stable. Or am I wrong in that assumption?

gskjold, 2022-10-31:
Not sure what device you have, but 2.1.11 is stable for most. If you have ESP32-S2 you could possibly experience unstability.

In any case you can wait for 2.1.12 which has optional power saving mode for WiFi which will fix instability. Working on releasing it right now.

shadowano, 2022-10-31:
I have a POW-U ESP32-S2. So if I understand correctly, 2.1.8 has wifi on all the time, and the problem on some devices with 2.1.11 is that sleeping wifi is the problem?

gskjold, 2022-10-31:
If 2.1.8 works for you, then 2.1.11 would work just as well. The issue have been the default "Max" powersaving enabled from 2.1.8 through 2.1.11. I have just now release 2.1.12 where power saving can be changed in WiFi config.

shadowano, 2022-10-31:
Ah, thanks for clarification :) will update to 2.1.12 later then

shadowano, 2022-10-31:
I'll leave this open as for documentation improvement

closed 2023-01-23 by gskjold