UtilitechAS/amsreader-firmware #349: Possible bug caused by clock ajustment, Daylight Saving Time (DST)

tsh1956, 2022-10-31:
**Describe the bug**
Possible bug caused by clock ajustment, Daylight Saving Time (DST)

**To Reproduce**
Observed at end-of day, the 30th of October. ('the 25-hour day')
No attempt to reproduce 

**Expected behavior**
Normal cap of hourly power consumption

**Screenshots**
[image]

**Hardware information:**
 - Norway
 -  Aidon
 - Encryption enabled no
 - AMS reader: Pow-U
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.1.11
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: defaults
 - HAN baud and parity: 2400 8E1
 - Temperature sensors no
 - ENTSO-E API enabled: no

gskjold, 2022-10-31:
Indeed, I see that the same happened to me... Will look into

gskjold, 2023-03-30:
Found the problem!