UtilitechAS/amsreader-firmware #350: Way to show raw received data?

sirtet, 2022-10-31:
**Describe your problem**
Something is off in my config, the AMS main page reports way too high daily energy export values, like an order of magnitude.
But the current export power seems to be correct or at least closer...
So i wonder if i can somewhere see the full received data.
Maybe in the debug?
Showing not only the raw bytes, but 
subsequent interpreted data? Is there human readable data of m-bus packets?
Or on the DLMS "Layer"?
Or the "raw" OBIS values?

The closest thing i can see the output of `mosquitto_sub -h localhost -t ams1/# -d`
```
Client mosqsub|2800-raspberryp sending CONNECT
Client mosqsub|2800-raspberryp received CONNACK (0)
Client mosqsub|2800-raspberryp sending SUBSCRIBE (Mid: 1, Topic: ams1/#, QoS: 0)
Client mosqsub|2800-raspberryp received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|2800-raspberryp received PUBLISH (d0, q0, r1, m0, 'ams1/id', ... (17 bytes))
CC:50:E3:95:96:18
Client mosqsub|2800-raspberryp received PUBLISH (d0, q0, r1, m0, 'ams1/meter/import/reactive/accumulated', ... (6 bytes))
24.920
Client mosqsub|2800-raspberryp received PUBLISH (d0, q0, r1, m0, 'ams1/meter/import/active/accumulated', ... (8 bytes))
1752.740
Client mosqsub|2800-raspberryp received PUBLISH (d0, q0, r1, m0, 'ams1/meter/export/reactive/accumulated', ... (8 bytes))
1684.100
Client mosqsub|2800-raspberryp received PUBLISH (d0, q0, r1, m0, 'ams1/meter/export/active/accumulated', ... (7 bytes))
263.750
```
It's not very nicely formatted and as that is all a bit new to me, it is a bit overwhelming.
A list of received OBIS numbers and their values would be easyer to compare against the meter's display.

Besides the general question about debug options:
In my case, the meter shows 1.8.0=25kWh, 2.8.0=175kWh. That corresponds to the first 2 readings in MQTT. So, the data mapping seems to be wrong. How to fix that? Hmm, should maybe be a separate issue?

**Hardware information:**
 - Country: CH
 - Meter: Omnipower
 - Encryption enabled yes
 - AMS reader: ESP32

**Relevant firmware information:**
 - Version: 2.1.11
 - MQTT: yes
 - MQTT payload type: raw

sirtet, 2022-10-31:
Yes, i did use that blogpost to establish the initial connection.
How would i use that RAW data on Telnet to see which data is being read?
```

* Please type the command and press enter to execute.(? or h for this help)
***
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop)(C1) Used 1 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /debugging.html over http...
(D) (loop)(C1) Used 1 ms to update temperature
(D) (AmsWebServer)Handling save method from http
(D) (AmsWebServer)Received Debug config
(I) (AmsWebServer)Saving configuration now...
(I) (AmsWebServer)Successfully saved.
(I) (AmsWebServer)Saving configuration now...
(I) (AmsWebServer)Successfully saved.
(D) (AmsWebServer)Serving /index.html over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /dayplot.json over http...
(D) (AmsWebServer)Serving /monthplot.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(V) (unwrapData)(C1) HDLC frame:
(V) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D
(V) 45 02 0C 84  FD 82 01 D0  30 00 00 8C  E5 D3 E5 72
(V) 38 33 19 39  A6 93 6B C0  0E C0 DA 9C  34 8C 7F EA
(V) EC 40 89 AF  0D 6C 05 31  30 27 DB F2  2B E1 D5 13
(V) 48 B5 CB EA  D8 8E 8F AC  79 E8 8F 9B  2F 85 5C 0F
(V) 98 97 74 56  0D F7 5C B9  F7 F4 C0 BF  BB 35 6D AF
(V) 54 63 E3 CE  4A 43 84 71  0B 12 D5 07  01 96 91 13
(V) DE DE 33 3E  5D F3 F1 D6  00 F1 F0 E8  E4 1F D8 AD
(V) D3 D5 0F 51  61 1E 81 A2  FB C2 63 CE  FA A5 D7 C3
(V) EC F6 67 68  2C 74 FB 96  E0 DD 54 44  0E C9 BB E2
(V) 71 70 1A 2A  A5 65 5E 0F  01 8B 68 F0  5D 0F 71 4A
(V) F8 6A B7 FD  D4 9D E1 40  2E 0A 80 A3  FE BA 3C 7A
(V) 72 13 38 EE  07 EC F6 5C  1A 15 24 9D  3A 6D F9 07
(V) A8 84 20 69  64 50 D0 0A  D1 70 EA 59  E9 3D 68 7B
(V) 4B F6 EF B8  12 5D C7 BA  BC 01 43 AF  DE FE FC 14
(V) 99 7C F0 A9  AB AF CE 4E  26 78 56 D6  02 7A 17 05
(V) 03 C2 13 FB  EA 77 88 F7  E3 38 05 76  A5 ED BF 49
(V) 6B 95 29 86  CE F2 E0 E5  AC 01 3E 6D  96 4D 8D B7
(V) A8 F9 73 6C  3E 49 00 60  7A 7E 6E F9  30 94 EB F4
(V) 01 3E 9F 03  BF 02 55 8C  B1 AB 41 1D  CA 5C BB 89
(V) 90 A1 94 2C  EE 61 95 14  5C CF 8A 19  3B 3C 62 C5
(V) 9F 27 9B 42  AD D9 55 93  45 C0 82 B3  19 80 39 D2
(V) 02 16 B5 6F  EF 5D 82 0D  54 95 6F E5  86 46 C3 54
(V) 01 0A 5C EA  BA 27 27 0D  09 EA 16 C2  62 FB 70 AC
(V) 89 C0 D5 8F  D2 68 90 7E  2D 7E 2C AF  83 26 D8 58
(V) 35 58 EB 4D  E9 17 18 17  00 C9 D6 CE  4D 81 12 A8
(V) 4C A2 D2 FA  E2 7A CD 20  23 3D 53 79  6A 2B 7F 5A
(V) D4 A7 89 AB  B2 94 65 A1  F5 A5 6B 85  B9 93 4A 6E
(V) 39 3E 35 3C  D2 5C 64 59  F8 70 48 2D  E1 42 48 16
(V) 86 EE B7 39  55 16 94 4A  8B EC EE 1C  18 CC 39 8A
(V) 6D E4 FD 20  85 63 28 5A  C8 13 7E
(V) (unwrapData)(C1) LLC frame:
(V) E6 E7 00 DB  08 4B 41 4D  45 02 0C 84  FD 82 01 D0
(V) 30 00 00 8C  E5 D3 E5 72  38 33 19 39  A6 93 6B C0
(V) 0E C0 DA 9C  34 8C 7F EA  EC 40 89 AF  0D 6C 05 31
(V) 30 27 DB F2  2B E1 D5 13  48 B5 CB EA  D8 8E 8F AC
(V) 79 E8 8F 9B  2F 85 5C 0F  98 97 74 56  0D F7 5C B9
(V) F7 F4 C0 BF  BB 35 6D AF  54 63 E3 CE  4A 43 84 71
(V) 0B 12 D5 07  01 96 91 13  DE DE 33 3E  5D F3 F1 D6
(V) 00 F1 F0 E8  E4 1F D8 AD  D3 D5 0F 51  61 1E 81 A2
(V) FB C2 63 CE  FA A5 D7 C3  EC F6 67 68  2C 74 FB 96
(V) E0 DD 54 44  0E C9 BB E2  71 70 1A 2A  A5 65 5E 0F
(V) 01 8B 68 F0  5D 0F 71 4A  F8 6A B7 FD  D4 9D E1 40
[... 106 lines cut ...]
(V) FF 06 00 00  2A E6 09 06  01 01 15 08  00 FF 06 00
(V) 00 EE 97 09  06 01 01 29  08 00 FF 06  00 01 6B E4
(V) 09 06 01 01  3D 08 00 FF  06 00 00 4A  E6
(V) (EnergyAccounting) Adding 0.0015 kWh export
(V) (EnergyAccounting)  calculating threshold, currently at 2
(V) (EnergyAccounting)  new threshold 2
(D) (loop)(C1) Used 1 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
```

Something additional that i did notice:
In the METER config, only MFG is populated, MODEL & ID are empty. 
But i guess that's not important.

Where/how is the raw data mapped to the MQTT topics (which seem to be mixed up in my case)? 
I would expect that's done via the OBIS codes that i assume are part of the DLMS Data?

Another possible inconsistency i see is that the wiki states Kamstrup meters normally send a list every 10s and one every hour.
My utility says they send the DLMS Push 1 List, which is different from the (normally used?) 

Screenshots from Kamstrup Document 5512-2584:
[image]

[image]

ArnieO, 2022-10-31:
Sorry, but that is a too big question to raise here. 
If you really want to dig into this you'll need to study all the documentation available from different sources, we cannot provide a lecture on all of that. 

All I can say from your screenshot is that you're receiving a valid datagram.

gskjold, 2022-11-01:
Resolved via email

closed 2022-11-01 by gskjold