UtilitechAS/amsreader-firmware #351: Need help or tips related to logging

tronde-ams, 2022-11-03:
Aidon 3p IT / Norway
ESP32 configured as Lolin D32
FW 2.1.12
No sensors
MQTT RAW active
Serial debugging / verbose

I am still experiencing strange things with prices over MQTT, but can't  tell for sure whether it is a bug or I am chasing ghosts (aka server trouble).
This time I am using test.mosquitto.org which have been stable for long time as I can see it.

Is it possible to see from the log what is pushed to MQTT? 

I have the text log, but it does not make any sense for me regarding this problem.

The problem is that the prices sometimes are rotated in an unexpected way between each hourly change. I have set up a screen capture at 5 fps rate to spot when changes occur. This is from today, and I can't twist my head to see how the server could do it?

[image]

mikkle, 2022-11-03:
Easiest way is to monitor what gets published to the mqtt broker from ams2mqtt.
I use a small python script for this kind of snooping:
https://gist.github.com/mikkle/eed4a89cd9d226b1f97ec74c8b62b086

Change the configuration in the script (lines 12-14) to fit your needs.
use "Publish topic" from your ams2mqtt MQTT configuration as root_topic.
The script will subscribe to this and all "sub-topics" and dump messages to the screen as they arrive.
When stopped (ctrl-c (really, SIGINT or SIGTERM)) it will dump a list of all topics it has received messages on.

The script needs paho-mqtt to run.
```
pip install paho-mqtt
python mqtt_mon.py
```
Feel free to steal/reuse/rework/etc
No support beyond the description above, though :)

example output:
[image]

Cheers
Mikkle

tronde-ams, 2022-11-04:
Script is up and running by now after fixing one minor typo in line 54 which should read
`client = mqtt.Client(client_id=f"{root_topic}_snooper")`

(Missing "p" in topic)

`python mqtt_mon.py > mqtt_log.txt `
will save to text file mqtt_log.txt instead of screen for those unfamiliar with oldfashioned DOS.

tronde-ams, 2022-11-06:
Logging works as expected. Just waiting for the error to re-appear.

closed 2022-11-06 by tronde-ams

mikkle, 2022-11-06:
> Script is up and running by now after fixing one minor typo in line 54 which should read

Cheers, nice catch - I've fixed the gist just in case anyone else stumbles upon it.
Hope you track down the weirdness! :-)

tronde-ams, 2022-11-07:
@gskjold 
I am certain something is wrong in the code.
This is from yesterday.

 
[image]
.
.
.
Detail:
I don't think I buy the idea of the broker making these changes when the max and min prices also changes and follows the other incorrect prices.  I understand that max and min are form 0h to 23h?
[image]

I have a complete log from the @mikkle  script for the period shown.
I have also seen some weirdness in the log from today, but I don't have time to dive into it now.

ArnieO, 2022-11-08:
Thank you @tronde-ams ! This sure looks weird. 
And... why does the date 2022-11-02 show up here...?
[image]

ArnieO, 2022-11-08:
Yes, something weird here. I don't think many are using the MQTT Raw format, so maybe not debugged.
I think we'll just have to wait for @gskjold finding time to look into this, I know he currently has a busy period at work. 

I read out here the values coming from MQTT Raw and made a graph in Excel to compare with the graph in the GUI, and it does not really make much sense:
[image]

tronde-ams, 2022-11-08:
Good to know I'm not chasing ghosts. I don't think many people will notice errors in prices if they don't look for them. They change often, and are just a number among many.

ArnieO, 2022-11-08:
Oh, I think many of us follow the prices quite closely.
But I do not think the Raw MQTT format is much used.

tronde-ams, 2022-11-08:
I have more....
Don't know whether they are related or not, but I think there is a connection.

I have three units running, all configured as Lolin D32.
One had FW 2.1.7 and is directly powered from external supply. No sensors, no MQTT and no logging.  Runs for months if I don't upgrade it or cut power.

On Nov 2, it started to mess up prices, probably just after it fetched new prices.
My two units running FW 2.1.12 was OK. I show one of them here.

I did restart the 2.1.7 in hope of it fetching new prices. Same prices. I tried to disable and reenable ENTSO-E. Same prices.
Then I updated it to 2.1.12. Same error. I have lost the screenshot, Then I tried to disable and reenable ENTSO-E once more. Same again. Then I tried to change price region. When I eventually tried the Swiss region, it loaded prices, probably the Swiss. Then I changed back to NO1 which it fetched correctly and as far as I know has done since then.

The corruption of prices in the WEB GUI reminds of something I reported about several months ago, namely that the modules seemed to self destruct after a while. It was related to prices in WEB GUI back then as well.

Why do the prices remain after both a restart and a FW upgrade? Is there a need to save them in flash?

Is there a possibility that we face some kind of memory corruption here, or that prices are written outside designated memory space? 

[image]

Prices from ETSO-E

[image]

ArnieO, 2022-11-09:
Nice catch again! 

Prices are read from Entso-E once per day and stored in flash memory. At reeboot it checks if stored prices are valid (to cover case of a long period without power or internet connection), and reads from Entso-E only if data is obsolete. I think this is in general a good strategy, but the consequence is that a reboot or upgrade will not force a new price fetch.

You could force a re-fetch of prices by doing a Factory reset, where all flash is deleted. You would then need to reconfigure the device again.

Very interesting error you're showing here - which @gskjold will surely look into when he finds time. And as you say: Potentially related to the issue on the MQTT "Raw (full)" payload.

gskjold, 2022-11-09:
Interesting stuff. Just a quick reply for now.

> Why do the prices remain after both a restart and a FW upgrade? 

They shouldn't. This suggests there is something fishy with the parser.

> Is there a need to save them in flash?

No, not saved to flash at all, kept in runtime memory.

> Potentially related to the issue on the MQTT "Raw (full)" payload.

Doubt it, there is no connection between the price storage and the MQTT handlers, but will look into.

Just so you are all aware, there is absolutely no sanity checks when reading prices from entsoe, so if the request returns more than 24 hours of prices, all bets are off. I know this was an issue earlier, not sure if that has returned now or what.

I can try to set up a automated check that verifies energyprice.json on fixed interval to monitor my own device for a while.

gskjold, 2022-11-09:
@tronde-ams Looking through your screenshots where you have 2-3 screenshots within the same hour. Can you actually see that the device is sending these values? It should only send prices when price change for current hour (on top of the hour) and should not send any price information for the rest of the hour.

gskjold, 2022-11-09:
Interesting log... Although I don't understand why it sends more often than once an hour. Will look closer at this