UtilitechAS/amsreader-firmware #352: Wifi "Power Save" off caused daily usage to be zero-ed every second

large, 2022-11-07:
I just installed the AMS meter and it worked great!
Connected to an Aidon meter, with only a RJ45 cable.

I tried to turn off the "Power saver" under Wifi settings and the daily hour/day usage was reset every second.
Turned it back to "minimum" and everything works fine.
Are this a known failure?

gskjold, 2022-11-07:
I have not tested this, but could you check if 2.1.13 fixes your issue?

And by the way, disabling power saving for Pow-U requires that USB power is always connected.

ArnieO, 2022-11-07:
I assume this is a Pow-U. It needs power saving to avoid it from loosing voltage and restarting quite often - which can potentially mess up some measurements - especially things like this running estimate of energy usage. 

Depending on your Wifi environment, the "Minimum" setting could be sufficient. Setting it to "Maximum" is necessary in some Wifi environment - with the drawback of some reduced responsiveness of the GUI. 

General advise: 

1. Try "Minimum", and if the device does not reset regularly: Keep it there. 
2. Switch to "Maximum" if needed due to frequent reboots.