UtilitechAS/amsreader-firmware #353: Unstable every now and then

shadowano, 2022-11-07:
**Describe the bug**
By looking at the up time in the user interface I see that the firmware can run stable for up to 3 days at best, but for other days it restarts more often. By looking at the "current hour used" in Home Assistant over time I see multiple restarts. See attached screenshot. Every black dots shows a restart (not at the hour).

I remember having more days up time with version 2.1.8, but I don't have any data for this version.

I'm going to use the Pow-U to send data to my HA server and set up a smart system. If it's restarting this often, this will make the system very unstable. Are there any logs we can look at, or find out why it's restarting this often?

Wifi is set to Minimum power saving. Not sure if I can test other settings for my hardware?

**To Reproduce**

**Expected behavior**
Firmware to be more stable and have weeks of uptime (at least)

**Screenshots**
[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: Pow-U, ESP32-S2
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.1.12
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8E1
 - Temperature sensors:
 - ENTSO-E API enabled: yes

**Additional context**

gskjold, 2022-11-07:
Maximum power saving is sometimes necessary for pow-u, which was default in 2.1.8. All devices have defaulted to minimum when upgrading to 2.1.12. On top of this, 2.1.12 have a bug where maximum power saving is not working. Please upgrade to 2.1.13 and try maximum power saving.

shadowano, 2022-11-07:
> Maximum power saving is sometimes necessary for pow-u, which was default in 2.1.8. All devices have defaulted to minimum when upgrading to 2.1.12. On top of this, 2.1.12 have a bug where maximum power saving is not working. Please upgrade to 2.1.13 and try maximum power saving.

I see, thanks, I'll try that and report back :) 

btw, thanks a lot for your quick reply, and all your effort on this awesome firmware with lots of feature:) Keep the great work up, it's really appreciated!

ArnieO, 2022-11-07:
More info here: https://amsleser.no/blog/post/25-pow-u-and-power-saving

shadowano, 2022-11-19:
If VVC is the reason it will help to connect it to power, right? Not sure if VVC is sent to HA, but will double check when I'm home again

ArnieO, 2022-11-19:
Yes - if this is the issue, connecting external power will cure the problem. 
For more explanation regarding this issue, please see chapter 5 of Pow-U user manual, downloadable from the product page on amsleser.no.

shadowano, 2022-11-23:
Doesn't look like the VCC is dropping below 3.1, IF MQTT data is sent when the VSS drops. I guess the IF is important to know when looking in Home Assistant :) 
[image]

ArnieO, 2022-11-24:
Vcc dropping should will be relatively slow, and reporting to MQTT is done each time the meter reports (each 2.5 or 10 second, depending on the meter). So all measurements should go to Home Assistant, then there is of course the question if HA stores and displays all values or does some averaging before making the graph (other users here might know?).
Was that a reboot at around 5:00 pm?
I cannot rule out an electronic malfunction in your device (e.g. too slow energy harvesting from M-bus). 
I'd be happy to swap your board with a new one to see if it helps; please contact me: post@amsleser.no.

bardahlm, 2022-11-25:
> So all measurements should go to Home Assistant, then there is of course the question if HA stores and displays all values or does some averaging before making the graph (other users here might know?). 

You can adjust the time interval to a short one to get the most details.

[image]

shadowano, 2022-11-29:
Sorry for late response, and thanks @ArnieO for your support and help. 
The last restart/crash the firmware/hardware did was 11 hours ago. I identified the value being sent to HA was 0 at 00:53:41 (included +1 hour for Norway):

`state: "0.0"
date: "2022-11-28T23:53:41.831694+00:00"`

At 00:53:37 HA shows that the VCC dropped to only 3.25. 
I do also find two places last 7 days where VCC dropped to 2.95 and 2.92. Other drops are not below 3, but two are close at 3.07 and 3.09. 
[image]

I have connected power to the Pow-U today, and will let it run for some time to just verify nothing else is causing this.

ArnieO, 2022-11-29:
> I have connected power to the Pow-U today, and will let it run for some time to just verify nothing else is causing this.

Looking forward to the update! Vcc drops to around 2.9V with Power saving set to Maximum is a quite clear indication it has rebooted due to voltage drop. 

Kindly run with power a few days, then contact me at post@amsleser.no to arrange a replacement.

shadowano, 2022-12-01:
Here is a graph before and after connecting power to the Pow-U, and it's easy to spot when that happened:) Is this an indication of "bad" hardware, @ArnieO?

[image]

ArnieO, 2022-12-02:
> and it's easy to spot when that happened:)

Wild guess: 🤣 
[image]

Thank you very much - that graph very interesting, there is a lot of information here. 

First, this indicates that the voltage drops you see is somehow linked to the energy harvesting from M-bus not being sufficient to maintain Vcc when device is powered from M-bus. 

But there is more to it: I struggle to understand the voltage steps. There is a huge supercapacitor (1 F) to keep Vcc stable. Huge capacitor and voltage steps is a strange combination. 

- What can cause the voltage to make steps (in both directions)?
- If the DC/DC converter struggles to harvest enough energy from the bus - how is it able to rapidly lift the voltage and keep it stable on a different level?
- And those rapid Vcc deep dives where it apparently reboots: Also strange.

I agree, there must be some kind of hardware issue here; could be a consumption issue in the ESP, could be a supercap that is in some way faulty, could be an issue with the buck converter controller... Difficult to say.

@shadowano: Please contact me on post@amsleser.no so that I can identify you and send you a replacement board!

shadowano, 2022-12-12:
After receiving the new unit, and importing the configuration (without secrets as that didn't work), I see the MQTT publish topic has some weird characters. I don't remember what I had, but I believe it was /data in the end and nothing else. 

[image]

gskjold, 2022-12-12:
Thanks for reporting, I will check if I am able to reproduce this. The config file is human readable, so you should be able to open it in ex. notepad. Could you check if there are more options than publish topic that are affected?

shadowano, 2022-12-13:
It is correctly stored as

> mqttPublishTopic ams/reader/data

in the config file. Screenshot from Notepad++ for reference. I haven't seen any other issues after importing.

[image]

closed 2023-01-05 by gskjold