UtilitechAS/amsreader-firmware #354: Firmware upgrade available to unauthorized user

jaboge, 2022-11-08:
**Describe the bug**
While I have enabled web security (for configuration - not everything), the firmware upgrade link is still available for anyone click.

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure your device is **not** at the latest version
2. Configuration -> Web -> Enable security: "For configuration"
3. **Do not** log in. Scroll down to the bottom of the page
4. The upgrade link is now visible. If you click it, you are warned not to proceed without connecting USB power.

**Expected behavior**
I would expect the upgrade functionality to be hidden from an unauthorized user while "configuration" is protected. This seems particularly important as an upgrade can/will brick the device while USB power is not connected.

**Hardware information:**
 - AMS reader: [Pow-U]

**Relevant firmware information:**
 - Version: [e.g. 2.1.11]

ArnieO, 2022-11-08:
Nice catch - thank you for reporting it! 👍

gskjold, 2022-11-08:
The upgrade would still require login, but I see your point. Will fix in the future

jaboge, 2022-11-09:
Oh - I aborted my test before it came to that. Then I gues this isn't really a big deal afterall.
If you do choose to hide the upgrade link, I think it's still a good idea to announce the new version to the guest user.

ArnieO, 2022-11-09:
> If you do choose to hide the upgrade link, I think it's still a good idea to announce the new version to the guest user.

I agree!

gskjold, 2022-12-09:
I cant find an elegant way to detect this unfortunately, so I'm not spending more time on it. The most important thing is that the upgrade endpoint is in fact protected

closed 2022-12-09 by gskjold