UtilitechAS/amsreader-firmware #357: When enegyprice becomes extremely low the price graph disappears

tronde-ams, 2022-11-11:
ESP32 / NO1

At 02:00 this night the price in NO1 became 0.00008 NOK/kWh. Then the price graph disappeared from WEB GUI for some hours when the current price remained low. MQTT still pushed prices, and /energyprice.json was OK.

I tried to change the multiplier to 10, and then the graph re-appeared as shown in the screen shot. I tried different multiplers, and the show / no show was somewhere between 6 and 7.

Maybe also check for negative prices?

[image]

ArnieO, 2022-11-11:
Even negative prices seems to work correctly here (NO3):
[image]

tronde-ams, 2022-11-11:
All of my three units behaved the same, but it is important to notice that the graph was OK until the price for the current hour (price/0) became extremely low. Until then, it showed the very low forward prices as it should, just like your graph also does.

The EUR price was 0.08/MWh which gives about 0.0008 NOK. When multiplied by 7 (0.0056) the graph re-appeared. When multiplied by 6 (0.0048) it was not shown).

It can be you will see it at 23:00 tonight or 08:00 tomorrow.

ArnieO, 2022-11-11:
Huh, sounds like a snag in the framework that "paints" the graphics.

tronde-ams, 2022-11-11:
price/0 just now is 0.0029. No graph. If I increase the multiplier to 1.719 so that price/0 = 0.005 the graph is shown.

gskjold, 2022-11-12:
Thanks, good catch, I see what it is, will fix for next release

gskjold, 2022-11-23:
Will be fixed in upcoming v2.2

closed 2022-11-23 by gskjold