UtilitechAS/amsreader-firmware #358: Pow-K unexpectedly unavailable on WIFI

henriklund, 2022-11-11:
**Describe your problem**
I have experienced twice in the past days that the POW-K dropped completely off the network (ie. not pingable / WebGUI down). It has happened at midnight (almost to the second) two days apart.

First time:
- checked LED (every 10sec blue followed by green flash -> device would seem to work)
- checked network (other Wif devices were online -> network & Wifi working)
- unplugged Pow-K, waited 5 mins, reinserted (no change to LED, device still not reachable)
- Forced Pow-K into access node: reentered name + fixed IP -> device rebooted -> not reachable
- Forced Pow-K into access node: reentered name + DHCP IP -> device rebooted -> device not reachable
- rebooted Wifi router -> device reachable
- during daytime : upgraded to 2.1.14

Second time:
- rebooted Wifi router -> Pow-K rebooted spontaneously (guess: due to prolonged time - 5 mins - when it was unable reachable MQTT during Wifi reboot)
- after reboot Pow-K was available again

Both times all other devices were reachable on Wifi. Wifi Router had not been updated for the past two months and uptime on the Pow-K before this was 6 weeks.

Question is thus if this behaviour has been experienced by anybody else.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: yes]
 - AMS reader: Pow-K
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - Version: 2.1.7 / 2.1.14
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: GPIO3
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: N/A
 - ENTSO-E API enabled: no

ArnieO, 2022-11-11:
Not important for your issue, but is HAN: GPIO3 correct? I cannot remember having used that GPIO even for the oldest versions Pow-K with ESP8266..?
Is your device an "old" Pow-K with ESP8266 or a newer Pow-K+ with ESP32?

Difficult to say what is the reason. I have seen the same thing happen in my network, where reboot of Wifi router restored connection issues. I've seen it happen (once, I think) with my Pow-K+ (ESP32), and I've also had a very similar problem with a Shelly Pro 4PM (also uses an ESP32).

henriklund, 2022-11-11:
Hi @ArnieO,

The device is the old ESP8266 (from last year). Setting are:
[image]

And even and old config backup from March shows the same:
[image]

ArnieO, 2022-11-11:
Ah... yes of course - UART0 is GPIO3! 
Sorry for that - my bad.

Regarding your issue: My guess would be that it is related to the Wifi network, maybe combined with some issue (design issue?) in the Wifi part of the ESP.

It is not uncommon to set up wifi routers in such a way that they regularly reboot, many of them can be set up to do it automatically at nighttime.

gskjold, 2022-11-12:
What is power saving set to in Configuration -> Meter? I would suggest playing with this setting and see if it gets better. Minimum should be good for pow-k, but you could also try Default and see if that changes anything.

On Main dashboard, what numbers are you seeing on RSSI?

Regarding reboot while offline, the firmware is set up to reboot if it is not able to connect to wifi after a certain amount of time.

ArnieO, 2022-11-12:
> Regarding reboot while offline, the firmware is set up to reboot if it is not able to connect to wifi after a certain amount of time.

The few times I have seen this (before the "Power saving" setting was introduced), the device did not know that it was not reachable over Wifi (no error blinking). I believe it was my Wifi router system that somehow failed.

henriklund, 2022-11-12:
> What is power saving set to in Configuration -> Meter? I would suggest playing with this setting and see if it gets better. Minimum should be good for pow-k, but you could also try Default and see if that changes anything.

First restart was on 2.1.7. AFAIK power saving was not available then. For 2.1.14, it is on 'OFF'

> On Main dashboard, what numbers are you seeing on RSSI?

It varies a bit -51 / -55 / -58 dBm. Q: what is 'power' setting in the Wifi settings (currently at 20,5dBm)?

> Regarding reboot while offline, the firmware is set up to reboot if it is not able to connect to wifi after a certain amount of time.

My router takes a few mins for the reboot to complete. Would it be an idea to permit this threshold to be tweakable?

henriklund, 2022-11-12:
> As far as I have seen, the ESP8266 does not react to that setting (no detectable change in power consumption).

The reboot of my router (which takes a some mins to complete) almost always results in the Pow-K to reboot (I measure on the uptime and get a notification when this gets reset). The fact that this is defined by the firmware is a very good explanation to this observation.

> the device did not know that it was not reachable over Wifi (no error blinking). My hypothesis is that it was my Wifi router system that somehow failed.

Exactly the same was what I saw...blue when receiving data from meter, followed by the green flash. No indication whatsoever that the device had lost Wifi connectivity. What is puzzling is that while the Pow-K is unable to connect to the Wifi network (absent in the MQTT server / not pingable) eventhough all other Wifi devices are available, it is able to detect it has lost Wifi (when router is rebooted) and thus opts for a reboot.

gskjold, 2022-11-14:
> First restart was on 2.1.7. AFAIK power saving was not available then. For 2.1.14, it is on 'OFF'

Try "Default", which would put it at exactly the same as it was on 2.1.7.

> My router takes a few mins for the reboot to complete. Would it be an idea to permit this threshold to be tweakable?

I've been thinking the same, so probably in the future.

henriklund, 2022-11-14:
Default appears to equal minimum :) Appart from the two oddities last week, the module has run without any restarts (except for the one a  few sec. ago forced by the change of Power Saving scheme). I'll keep my fingers crossed that this was 'once off' scenarios.

henriklund, 2022-11-29:
Past two weeks have been uneventful (ie. module avaiable). I will close this.