UtilitechAS/amsreader-firmware #359: Energy usage off by a factor of 10

TheYves, 2022-11-11:
Hi

The graph "Energy use last 24 hours" shows values which are off by a factor of 10. So for example, the value showed on the dashboard for one hour is 7.6 kWh whereas it really should be 0.76 kWh. I'm sure that 0.76 is correct because that's the value I have from my energy provider.  

The current "Import" value on the other hand, which is being updated every 10 seconds, shows correct values.

Thanks for your help.

**Hardware information:**
 - Country: Switzerland
 - Meter: Kamstrup
 - Encryption enabled
 - AMS reader: Pow-K+ HANmodule
 - M-bus adapter (if applicable): ?

**Relevant firmware information:**
 - Version: 2.1.14
 - MQTT: yes
 - MQTT payload type: Home Assistant / json
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400, 8N1
 - Temperature sensors: ?
 - ENTSO-E API enabled: ?

gskjold, 2022-11-11:
Adjust multiplier for accumulated accordingly in Config -> Meter -> Multipliers. Should be 0.1 in your case.

TheYves, 2022-11-11:
Thanks for your quick reply. Which one of these do I have to set? Accumulated?

gskjold, 2022-11-11:
Yes. Remember that already saved data for graph will not be corrected

TheYves, 2022-11-11:
Thank you, it works like a charm. Is there a way to clean up the old data?

gskjold, 2022-11-12:
Well, you could go to System -> Configuration file, dowload it, remove a digit from monthplot and dayplot and upload.

closed 2022-11-21 by gskjold