UtilitechAS/amsreader-firmware #361: Max usage graph is "off" / calculated oddly

drhousedk2, 2022-11-15:
I have my meter configured to 230 volt and main (input) fuse to 25A which results in a dial that goes to approx. 5 kWh (is my guess).

If I have three phases with 25A each my max. capacity if maxing all three phases should be 17.25 kW. As it is right now, charging my EV at 11 kW will make the gauge red and full, although it should show about 63% usage. Bar graph shows phase load correctly.

Is this just a config thing, or what am I missing?

[image]

tronde-ams, 2022-11-15:
Seems like you have configured your meter as 230V IT (Norwegian system). I guess so, because the graphics says "Voltage between". If so, you will max out at 9947W.

ArnieO, 2022-11-15:
I think @tronde-ams is right. You probably have a 400V distribution system. Go to Config->Meter and set:
[image]

drhousedk2, 2022-11-15:
You guys are absolutely correct. Changing to 400 volt made everything look a whole lot better. 🤩 Thank you!!!