UtilitechAS/amsreader-firmware #363: Low voltage on ESP after upgrading to 2.1.14

bottiger, 2022-11-18:
I had Pow-K+ running for months without any issue. I just discovered there was some outages in the data collection in home assistant, and decided to update the firmware to 2.1.14.

I can still access the web interface, but the ESP indicator is red and the interface displays a voltage of only 0.1V

I plugged a usb cable into the external power and did some basic measurements which seemed correct. However, the webinterface still said 0.1V and showed a red indicator for the ESP

See screenshots for details

Is is currently unclear to me if the low voltage problem was there before the firmware update

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: yes
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: 2.1.14

[image]

[image]

[image]

ArnieO, 2022-11-18:
There is unfortunately a layout error on your card (layout v1.5) - so it cannot read Vcc (I used a pin that cannot be configured as ADC). 

Fix: Go to System -> GPIO and delete the number in the "Vcc" box. When that is done, Vcc is not displayed and the "traffic light" will be green. 

When I shipped these cards, I configured them in that way, resulting in Vcc not being displayed. The upgrade to v 2.1.14 seems to have altered that setting. 😉 

PS:
Showing Vcc is not really very useful on Pow-K which has a stable input voltage from the meter. It is somewhat useful on Pow-U, that generates Vcc 3,3V from the M-bus line (24/12V) - and in some cases its Vcc voltage will vary, see https://amsleser.no/blog/post/25-pow-u-and-power-saving

bottiger, 2022-11-18:
That certainly turned the ESP back to green. Thank you. I'm still having problem connecting to the HAN port, but I have not spend time debugging that yet

ArnieO, 2022-11-18:
- Check that System / GPIO / HAN is correct. (GPIO16)
- Check that Config / Meter / Parity is correct (use link "known hardware configurations" in the menu)

bottiger, 2022-11-18:
That certainly fixed it! thank you again. HAN was not configured for GPIO16, but I don't recall what it was and I have no idea how it could have been changed. 

Have a nice day