UtilitechAS/amsreader-firmware #364: Daily accumulated price resets again at 01:00

tronde-ams, 2022-11-20:
ESP32 configured as Lolin D32. FW 2.1.12 and 2.1.14.
Aidon 3p IT / NO1.

The daily accumulated price resets as it should at midnight, but it resets again at 01:00 too.

In addition the montly accumulated price is reduced by the amount from the hour between 00:00 and 01:00. I have two units with 2.1.12 and one with 2.1.14 and all behaves the same way since I noticed it earlier this week. 

After 01:00 it seems to be OK.

[image]
[image]

gskjold, 2022-12-01:
I'm having a look at this now. When you say it is ok after 01:00, does that mean the day cost is correct? That it have even included cost between 00 and 01?

tronde-ams, 2022-12-01:
It resets at 00:00 as it should and starts counting both price and kWh until it resets again at 00:10. Then it counts both price and kWh as it should until 00:00. The implication of this is that the accumulated price and kWh between 00:00 and 01:00 is lost so only 23h of data is used.

For mee, it seems to be a mix of UTC and CET time?

gskjold, 2022-12-01:
> For mee, it seems to be a mix of UTC and CET time?

Effectively that what it looks like during CET. The same would happen in CEST though. 

Moving line 85 to above 78 here fixes the issue: https://github.com/gskjold/AmsToMqttBridge/blob/1ef5703971ab88ac65575842ac127518a24acb31/src/EnergyAccounting.cpp#L85
https://github.com/gskjold/AmsToMqttBridge/blob/1ef5703971ab88ac65575842ac127518a24acb31/src/EnergyAccounting.cpp#L79

closed 2022-12-01 by gskjold

gskjold, 2022-12-06:
v2.2 beta out for testing now:
https://github.com/gskjold/AmsToMqttBridge/discussions/370