UtilitechAS/amsreader-firmware #365: Wired AmsToMqttBridge?

attah, 2022-11-27:
**Is your feature request related to a problem? Please describe.**
There are some places without wifi, but that may well have wired ethernet (e.g. basements in multi-tenant houses).
And there are even some nice PoE-powered, wired, ESP32s like https://shop.m5stack.com/products/esp32-ethernet-unit-with-poe?variant=42675684966657

**Describe the solution you'd like**
Would you be interested in enabling AmsToMqttBridge to use the built-in ethernet MAC on the ESP32?
I'm currently a bit low on spare time so haven't got far, but i have bought the hardware, set up the environment and programmer, and am slowly starting to look at what it would take.

So i just wanted to gauge how, if at all, involved you'd want to be.
If you are anything like me with open projects, there might be some parts and/or some prep that you'd want to get out of the way first.
Or you maybe already thought of this and have some ideas/suggestions... or you might even want more involvement.
(I'll gladly sponsor you for the hardware)

Or if you don't, i hope you'd at least accept pull requests for this so that it doesn't end up being a fork.

What do you say? :)

**Describe alternatives you've considered**
One can always sprinkle wifi access points everywhere... but this is cheaper and options are always nice.

**Additional context**

ArnieO, 2022-11-29:
Thank you for interesting proposition and link!

I represent amsleser.no, and we have had several requests for AMS-readers with Ethernet/PoE connection option.

For the M5stack ESP32 PoE device: If you want to use the exposed UART port for the signal from the AMS meter, you'll need to add some interface hardware towards the meter to condition the incoming signal, depending on the physical interface on the meter. 

1. The simplest case would be a Kamstrup meter, if you connect to the 6-pin connector in the CCC bay; the serial signal there is from a 0/3,3V push/pull output.
2. The "Swedish/Dutch" P1 interface is a bit easier, but has an Open Collector output, which must be handled correctly, and preferably including inverting the signal.
3. More complex: Variants of M-bus-type signalling. (Norwegian "NVE standard" or European meters with higher signal voltage). The incoming signal must be scaled down from 24/12V or 32/18V (depending on interface) to 3,3/0V for a ESP32 GPIO. 

Then of course the firmware needs to be modified, as you indicate - to send data over Ethernet in stead of Wifi.

attah, 2022-11-29:
Very interesting to hear!

Are there any plans for a wired amsleser? 
The interfacing bits are indeed another reason for purpose-built hardware.
For where i want this, we have a P1 interface, so should be doable. From what i could find online the ESP32 has a setting for inverting the UART. Or do you consider that not usable for some reason?

For my own meter, i have a "Norwegian" version, and there the POW-U was a godsend since i would not manage to power anything from that with my limited electronics experience.

Speaking of which; PoE is probably overkill since there shouldn't be reason to run Ethernet and WiFi at the same time.
You are quite right in that i should look at the electrical part first. Otherwise, what will i even test with. :)

Should you consider making purpose-built hardware, consider me interested in 4 units.
And in the meantime i'll keep tinkering.

attah, 2022-11-30:
> For your P1 port situation, you could maybe consider sending data out via mobile phone network using an NB-IoT connected access point? Using Pow-P1 devices on the meters. Tidy install without all those cables.

Just for completeness (it's my problem after all); That would have been a good solution, except it is in 4 different buildings, but they all just happen to have Ethernet with PoE a meter away.

ArnieO, 2022-12-01:
> That would have been a good solution, except it is in 4 different buildings, but they all just happen to have Ethernet with PoE a meter away.

Ah, OK. In that case not a good solution. 
I had in mind one of those technical rooms in basements of commercial buildings with many meters in the same room.

bardahlm, 2022-12-02:
> Speaking of which; PoE is probably overkill since there shouldn't be reason to run Ethernet and WiFi at the same time.

Wifi is not the only reason for PoE. Another use-case is surveilance cameras. There are also cheap power injectors available.

I'd prefer PoE over a separate power supply.

Do ethernet use more power than wifi?

ArnieO, 2022-12-03:
> Do ethernet use more power than wifi?

I don’t know yet! 😄

ArnieO, 2022-12-03:
The ability to handle long cables is one of the few positive things I can say about the standard chosen for the Norwegian AMS meters: High voltage levels (24/12V) and moderate bit rate (2400 baud). 

The standard adopted by Sweden is a more modern standard. Data delivered from an Open Collector output, nominally powered at 5V and at 115 200 baud bit rate. Much easier to interface, but don't expect it to handle more than a few meters of cable.

ThorAlex87, 2023-01-12:
I'll add my vote fore a wired version! I'm moving to a new house soon and expect to have to buy a second AP to get good enough coverage for the reader. Or run an ethernet wire from the meter and put the reader in my livingroom...

PTRSC, 2023-01-26:
I would also like to be able to run it on an ESP with Ethernet and PoE, for example Olimex ESP32 Power-over-Ethernet ISO

GEAG, 2023-08-11:
It is quite feasible to run an ethernet socket together with an ESP32, with the energy solely supplied trough a P1-DSMR version 5 port. P1-DSMR ports must be able to supply up to 1.25 W of power, according to ist public specification. There is no need for POE. We have built a bunch of such devices running already for several months with no major problems. We currently use a different firmware than AMSreader. You might want to look at the Wt32-ETH01 board (about 15 EUR). This still needs the additional P1's adaption to serial circuit and maybe an USB port for easy flashing and debugging.

ArnieO, 2023-08-14:
> It is quite feasible to run an ethernet socket together with an ESP32, with the energy solely supplied trough a P1-DSMR version 5 port. P1-DSMR ports must be able to supply up to 1.25 W of power, according to its public specification. There is no need for POE. We have built a bunch of such devices running already for several months with no major problems. We currently use a different firmware than AMSreader. You might want to look at the Wt32-ETH01 board (about 15 EUR). This still needs the additional P1's adaption to serial circuit and maybe an USB port for easy flashing and debugging.

Thank you for sharing your knowledge! 

The P1-DSMR port is clearly the less challenging regarding ethernet, as there is 1.25W power available. 
(Our Pow-P1 board is designed for P1-DSMR port; Wifi only.)

The challenge is our best selling boards:

**- Pow-U (for meters with M-bus interface)** 
There is no clear specification for this standard, the implementations vary between meter manufacturers, even the M-bus voltage level varies. And available power is generally not documented by the manufacturers.
For the meters used in Norway we have:
. Aidon: 700 mW
. Kaifa: 500 mW
. Landis & Gyr: We have not found documentation on available power.
M-bus meters used in Europe (Sagemcom etc): We have not found documentation on available power.

**- Pow-K (a plug-in board for Kamstrup Omnipower meters)**
Available power around 300 mW.

So here you see the reasons we have been thinking PoE, as there in general is significantly less power available on those devices compared to the 1.25 W available on a P1-DSMR port. 

Also complicating this a bit is the fact that the current consumption of such a device is varying a lot depending on what the device is doing. While transmitting on Wifi, the ESP32 current draw is substantially higher than what is allowed as maximum in the P1-DSMR specification. Some meter designs are more tolerant than what is said in the specification, which means that you are not necessarily compliant to the spec even if things do work correctly. We have seen this on one of the newest meter models (Kamstrup Omnia E-meter): It is less tolerant to short current peaks, so we recently found that our Pow-P1 design needs a modification to be able to run on that meter.

We do not know yet how peak current draw is on an ethernet system, and whether this could be a problem; just pointing out that this is an issue to take into account in addition to overall average consumption.

**EMC**
We are in the process of CE-marking our products. As part of that process, the designs will be submitted to an EMC test. In order to pass such an EMC test, an Ethernet socket will have to be sufficiently protected regarding EMC issues. The potential of a (very) long ethernet cable being connected makes the test regime more severe compared to our current design, where there is only a short cable between the reader and the power meter. 

The [Wt32-ETH01](https://www.seeedstudio.com/Ethernet-module-based-on-ESP32-series-WT32-ETH01-p-4736.html?queryID=033d8e8f5859cdbb445ca27e8328344a&objectID=4736&indexName=bazaar_retailer_products) from Seeed Studio does indeed look very interesting, and could be an interesting starting point for a design for P1-DSMR - however it seems to be inadequately documented. And it is does not seem to be CE marked, so we cannot expect it to hold up in an EMC test. (I also note that they do not seem to use original Espressif ESP32 module; making it even more difficult.) If you find more documentation on this board, please let me know!

Was it this board you have used for your implementations?

gskjold, 2023-08-27:
I have made this work with LilyGO T-Internet-POE. Does anyone out there have any other boards available for testing?

attah, 2023-08-27:
I still have those M5stack ones with PoE that i linked in the first post. (But i haven't really got any further than setting up the build environment - other projects taking my limited time).

gskjold, 2023-08-28:
No guarantee here, but I have added a hardware profile for M5 PoESP32 and corresponding ethernet config. Hopefully it works, and if you want to try it, here is the firmware:

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12452534/esp32.zip)

attah, 2023-08-28:
Excellent! Thank you!
I just need to A) find them again, and B) do the hardware side of things (perhaps it was just a pull-up, need to re-research that).

gskjold, 2023-08-31:
Sorry, I should have been more clear on how I (for now) have implemented this. You will still need to connect to the AMS2MQTT access point to set up the device. You will first get the hardware menu where you get to choose the hardware profile. Choose M5 PoESP32 and continue. You will then get the option to activate ethernet

attah, 2023-08-31:
Very nice! That seems to have worked - it shows up with hostname "tasmota" though.
Now i know what i'm doing this weekend :)

attah, 2023-09-02:
Success!
While i did mess with the settings, this should be back what it started with after selecting the correct config.
(UART2, pullup, inv)
I did end up needing an external pull-up though (10k to be at the safer end of normal values).

[image]

Wupsman, 2023-09-08:
Awesome what you share with us here 🍾🎉👍

I tried it right away and it works perfectly.

Only the description settings for Austria didn't work for me.

Cheers

[image]

ArnieO, 2023-09-09:
> Only the description settings for Austria didn't work for me.

Exactly what is it that does not work with this setting?

Wupsman, 2023-09-09:
I have a Smartmeter Sagemcom T-210-D from the grid operator EVN, Baud fits with 2400 but Party I had to set 8E1 and like the Kaifa MA309M I only needed the Encrypton key (as in the picture above) and also the format is MBUS.

ArnieO, 2023-09-09:
> I have a Smartmeter Sagemcom T-210-D from the grid operator EVN, Baud fits with 2400 but Party I had to set 8E1 and like the Kaifa MA309M I only needed the Encrypton key (as in the picture above) and also the format is MBUS.

The "Austria" settings are only for reading the electricity prices. Data format is linked to the type of meter, disregarding which country you are in. 2400 baud / 8N1 is normal for meters with M-bus output.

So you are using this with an Mbus-to-ethernet adapter?

Wupsman, 2023-09-10:
I can hardly believe what you have created with the AMS Reader, thank you so much.

I had tried to read the smart meter with the Python script from greenMikeEU, but it didn't work for me, it kept stopping. 

Then I tried an Olimex PoE + Farley TTL to MBUS with the currently released amsreader firmware and only the 8E1 setting worked. 

[image]

I then came across "Wired AmsToMqttBridge" here and tried the firmware posted by gskjold and this version is simply awesome on the M5 PoESP32 + Farley TTL to MBUS! 

[image]

 I also tried the 8N1 setting today and it works fine too, sorry for my misreport. 

[image]

What I would still like to know is, where can I find the firmware code with the options for darkmode and eth?

cherrs

gskjold, 2023-09-10:
Excellent, thank you for the positive feedback!

You can find the source in the dev-v2.3 branch: https://github.com/UtilitechAS/amsreader-firmware/tree/dev-v2.3
Just pushed it, so it is up to date now, including the ethernet implementation. 

Its rather rudimentary at this point: https://github.com/UtilitechAS/amsreader-firmware/blob/dev-v2.3/src/EthernetConnectionHandler.cpp

ArnieO, 2023-09-11:
@Wupsman : Thank you for your kind words!

You are maybe aware of this, but in case someone else from Austria finds this thread:

Our [Pow-U](https://lectronz.com/products/pow-u-han-port-reader-for-m-bus-meters-esp32) device with an RJ-12 cable plugged into its RJ45 connector directly reads the Austrian meters (also used in some other mid-European countries) that have M-bus signal on pins 3 and 4 of their RJ12 connector. 

(And I just realized we have not yet updated the description in the webshop to provide that information regarding ... so I have to fix that!)

But of cause - that will be a wifi connection from the meter - and not cabled Ethernet, which is what some of you following this thread are asking for.

egiljae, 2023-10-27:
I got a LilyGo T-Internet-POE board working over WiFi, but wired would be preferred. Will the M5 PoESP32 firmware posted here also work with the LilyGo board?

gskjold, 2023-10-28:
Yes, there is a profile for that board already, I have the same board.

markuspeter, 2023-10-29:
Hi,

> I did end up needing an external pull-up though (10k to be at the safer end of normal values).

Is that necessary to get it working ? Where exactly would I put that pull-up ?

I finally received a m5stack poesp32 and Farley TTL to mbus converter yesterday, managed to load the firmware from this thread and start up Asmreader, However, I get "No HAN data received for at least 30s" no matter what configuration I choose.

What options did you set under "Hardware"? I'm sure about UART2, but  what about the other fields and checkboxes ("pullup" for example) ?

I do get ~29V on the MBus dropping regularly to around 19/20V, therefore I guess the L&G 450 does output data. I connected the MBUS to the converter, and the converter to the Poesp32, using Rx (Rx pin on converter & poesp32 white cable).

For a brief moment, probably when auto-detect was initially activated, I've been getting a different message ("Unknown data received, check meter config") - but the UI doesn't let me put auto-detect back.

I was hoping to get some data back, since you @attah and @Wupsman were able to make it work with the same components.

Btw - awesome project, I'm blown away, even though I didn't get everything working yet!

attah, 2023-10-29:
Yes *needing* means it was necessary to have it.
Pulling up implies the data line up to something of higher voltage; i.e. +5V in this case. 
But i have the Swedish standard with TTL signals already available, though "open collector", i.e. needing to be pulled up.

If you are running a MBUS converter it should probably not need that (and whether you do would depend on the specific converter, and not be a general truth).

My post seems to say i have uart2, pullup and inv compared to with whatever is default. Fairly sure pullup does nothing, and i'm not sure inv does either. Inv(ert) sounds  like something you shouldn't need with a MBUS converter.

If you, as you say, only have connected one cable - that's your first problem. You also need ground, or there will be no way for the esp to see the signal.
I would have expected that it is the TX on the converter that should be connected to the RX of the PoESP.
But it definitely looks like the PoESP is mislabeled (aka labeled from the point of view of what you connect to it and not itself). Yellow is its data-in.
No idea how i figured that out previously, must have looked up the respective GPIO functions in my sleep or something.

If that still doesn't work; have you checked the baud rate? MBUS should be 2400 but the looks of it.

markuspeter, 2023-10-29:
Thanks for pointing out the RX/TX issue - now that you said it, it seems obvious that it's the yellow cable (there is a label "G16" on the other side).
Of course I also connected Gnd and 5V between the Poesp32 and the converter. 

Unfortunately, this still doesn't work, I still get no data.

I use the same converter as @Wupsman, the Harley TTL to MBUS converter. RX seems to be the correct pin, it's identical to the picture of above [post](https://github.com/UtilitechAS/amsreader-firmware/issues/365#issuecomment-1712878848)  (and in [this post](https://github.com/UtilitechAS/amsreader-firmware/issues/433#issuecomment-1734059616) using the same module).
Nevertheless, I tried the other pin as well, without any luck.

I tried 2400/8N1 as well as other settings. I assume I should get something even if the data were encrypted (which I don't know yet). Telnet debugging does not show any incoming data neither ("no HAN data received last 30sm single blink").

Running out of ideas what to try next, maybe I've got a faulty converter.

gskjold, 2023-11-10:
> @gskjold Is this mainlined yet BTW?

Not yet, working on some changes I want to include in 2.3 before releasing it for full testing.

gskjold, 2023-11-10:
Added profile for WT32-ETH01, based on schematics from @gellfuen in discussion #673. Untested, hope it works. Remember you need to preconfigure the board via WiFi AP before hooking it up to ethernet.
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13316824/esp32.zip)

gellfuen, 2023-11-10:
Many THX, module should arrive tomorrow, will start testing tomorrow. THX for the WiFi AP info!

weisstechas, 2023-11-10:
I just received a LilyGO T-ETH-Lite board, https://www.lilygo.cc/products/t-eth-lite. It uses an W5500 ethernet controller, and comes in a small package, even with the POE shield attached. Is it possible to add support for this board?

egiljae, 2023-11-10:
I just received a LilyGO T-ETH-Lite board, https://www.lilygo.cc/products/t-eth-lite. It uses an W5500 ethernet controller, and comes in a small package, even with the POE shield attached. Is it possible to add support for this board?

gellfuen, 2023-11-11:
Hello, (discussions #673) got the WT32-ETH01 today and was able to flash the firmware from esp32.zip from yesterday with serial connection.
Test output/steps:

1. Disconnected power, removed flash jumper and reconnected power.
2. Connected to the WiFi AP.
3. Selected profile for WT32-ETH01 and configured AP at first boot.
4. I let the RS232 converter connected and saw at serial terminal the IP the WT32 got from DHCP.
5. Next step was to switch at configuration the connection from WiFi to Ethernet.
6. Device performed a reboot.
-- At this point, the serial terminal output stopped!
-- So I looked at DHCP for a new MAC address and found a device called tasmota :-) - I do not have any tasmota devices.
-- I'm able to connect over that new IP to the WT32 again, but I often lost  the connection, what I mean here is that I lost pings.
-- If I try to change from DHCP to fixed IP, that change will not happen, but the configured values are stored on the webpage and I can still see my entered values even after a reboot.
-- if I check settings at status the network section is empty and the MAC addresses at device information are not the same I can see on my DHCP.
Tried to add a picture of that:
[image]

Do you need something from my devices?

BR

gskjold, 2023-11-15:
Thanks, I see there are some things I forgot to implement for wired, including network status and fixed IP :) Will have a look at it when I have available time

markuspeter, 2023-11-19:
> Running out of ideas what to try next, maybe I've got a faulty converter.

I finally received a new converter, and it is working now. Looks like I wrecked the first one by inverting the pins to the poesp32, or it was bad from the start.

This works great with the L&G E450 in Switzerland (provider Romande Energie) with above settings (2400baud 8N1), no key needed. I get readings every 5 seconds using MQTT with the HomeAssistant payload config, which works fine, except for some reason I get 1 reading per minute which is all empty/zero.

GEAG, 2023-11-19:
> This works great with the L&G E450 in Switzerland (provider Romande Energie) with above settings (2400baud 8N1), no key needed.

Swiss L&G E450 from RE by default use 8E1, not 8N1.

> except for some reason I get 1 reading per minute which is all empty/zero.

L&G E450 push different sets of data values (I call it a telegram) at different intervals. The 1/minute telegram does not contain the respective power value, but other data values like energy counters. AmsReader sets a 0, if the value (assumed to be always delivered) is not present in the telegram. That's a known bug.

Most other SM do not issue different telegrams at different intervals. Because of the low throughput at 2400 baud 8E1 and a low S/N ratio of the DLMS data encoding, the data is partioned into different telegrams. 

Some other SM use higher baud rates (up to 115200), 8N1 asynch coding (saves 10% bandwith) and may not repeat the constant COSEM meta-data in every telegram, which dosen't add anything useful but clogs up a very large part of the channel's capacity.

markuspeter, 2023-11-19:
> Swiss L&G E450 from RE by default use 8E1, not 8N1.

Just changed the configuration to 8E1, and this works as well. Not sure if there is a difference, from a user point of view I can't see any - is there ?

> L&G E450 push different sets of data values (I call it a telegram) at different intervals. The 1/minute telegram does not contain the respective power value, but other data values like energy [...]

Good to know, because I was looking through the open issues and didn't see any related to that. Thank you for the explanation. I guess I'll filter that on the Home Assistant side of things for now.

gskjold, 2023-11-19:
> AmsReader sets a 0, if the value (assumed to be always delivered) is not present in the telegram. That's a known bug.

I thought I had that covered by now :thinking: I should have a look at that again

gellfuen, 2023-11-21:
> Thanks, I see there are some things I forgot to implement for wired, including network status and fixed IP :) Will have a look at it when I have available time

Hello, no problem! I have time, I can wait until it's finished. BR

gellfuen, 2023-12-05:
Hello
just kindly asking for update about the missing stuff for wired profile for WT32-ETH01?
BR

gskjold, 2023-12-10:
Here is an update, fixing hostname and fixed IP, as well as IP display on status page:
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13628371/esp32.zip)

gskjold, 2023-12-10:
Btw, I may have changed some things, so back up config, erase flash before flashing new firmware and set it up from scratch, then load config

egiljae, 2023-12-11:
> @egiljae I have one of these actually. Have you ever made a arduino sketch for it? Looks like the ethernet implementation is slightly different

I haven't had a chance to code a arduino sketch yet, but I successfully set it up in ESPHome with the following config for ethernet:

`ethernet:
  type: W5500
  clk_pin: GPIO10
  mosi_pin: GPIO12
  miso_pin: GPIO11
  cs_pin: GPIO9
  interrupt_pin: GPIO13
  reset_pin: GPIO14
  clock_speed: 30`

gellfuen, 2023-12-12:
Hi, uploaded software from esp32.zip(10th December) but sorry no success:
ESP (WT32) is only cycling:

I can see that text at terminal:
...
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4
(I) AMS bridge started
(I) Voltage: 0.00V
(D) ESP32 LittleFS
(D)  size: 1441792
(I) No configuration, booting AP
(I) Entering setup mode
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4
(I) AMS bridge started
(I) Voltage: 0.00V
(D) ESP32 LittleFS
(D)  size: 1441792
(I) No configuration, booting AP
(I) Entering setup mode
ets Jun  8 2016 00:22:57
....
At first I erased the flash like you wrote and loaded your files. >> ESP is only cycling...
Then I saw, that in the zip file the file "bootloader_dio_40m.bin" was not in place like at the older zip file, so I copied that and flashed again, but the behavior at the ESP is the same :-(.

gellfuen, 2023-12-12:
... oh! still recognized "bootloader_dio_40m.bin" will not be flashed

gellfuen, 2024-01-07:
Hello, Happy new Year!
just kindly asking for update about the missing stuff for wired profile for WT32-ETH01....
BR

gellfuen, 2024-02-09:
@gskjold: I don't want to be a nuisance, but from your last statement it seemed that only one small thing was missing. Will that perhaps still be finished after all?
BR

gskjold, 2024-02-13:
Please try the attached version, it should work for you

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14271512/esp32.zip)

gskjold, 2024-02-14:
I see what is going on, try this one:
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14285070/esp32.zip)

gellfuen, 2024-02-18:
THX,
hmmmm still rebooting, but much slower...

Bootlog:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4
(I) AMS bridge started
(I) Voltage: 0.00V
(D) ESP32 LittleFS
(D)  size: 1441792, used: 8192
(I) No configuration, booting AP
(I) Entering setup mode
(I) (setupHanPort) Setting up HAN on pin 255 with baud 0 and parity 0
(D) Hardware serial
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40116938: ffffffff ffffffff c332c3a6
Core  1 register dump:
PC      : 0x4011693c  PS      : 0x00060330  A0      : 0x80117d06  A1      : 0x3ffcb6b0
A2      : 0x00000001  A3      : 0x00000009  A4      : 0x011a0009  A5      : 0x00000009
A6      : 0x00060323  A7      : 0x00060320  A8      : 0x80116eb5  A9      : 0x3ffcb690
A10     : 0x00000009  A11     : 0x00000011  A12     : 0x3ff49054  A13     : 0x0000abab
A14     : 0x003fffff  A15     : 0x00060320  SAR     : 0x00000009  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4008a64c  LEND    : 0x4008a657  LCOUNT  : 0xffffffff

Backtrace: 0x40116939:0x3ffcb6b0 0x40117d03:0x3ffcb6d0 0x401158c3:0x3ffcb700 0x40113170:0x3ffcb760 0x400dcdd5:0x3ffcb7b0 0x400dceb9:0x3ffcb7e0 0x400d483f:0x3ffcb800 0x40115b9d:0x3ffcbaf0

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4

regards

egiljae, 2024-02-20:
> Here is an update, fixing hostname and fixed IP, as well as IP display on status page: ~[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13628371/esp32.zip)~ edit: Found a fault in the firmware above, use this: [esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13628515/esp32.zip)

Trying to setup LilyGo T-Internet-POE with this firmware. I have previously been using UART2 (GPIO16) while in wireless mode. Ethernet mode works fine, but as soon as I select UART2 for HAN RX, the board becomes unresponsive. I guess the ethernet chip uses UART2? Which pin should I use for HAN RX on this board while in ethernet mode?

ArnieO, 2024-02-20:
> Trying to setup LilyGo T-Internet-POE with this firmware. I have previously been using UART2 (GPIO16) while in wireless mode. Ethernet mode works fine, but as soon as I select UART2 for HAN RX, the [...]

You need to study the documentation for your board, select a GPIO that is not used by the board, use that GPIO for HAN Rx - and adjust the configuration for the firmware to read HAN data from that GPIO.

egiljae, 2024-02-20:
> You need to study the documentation for your board, select a GPIO that is not used by the board, use that GPIO for HAN Rx - and adjust the configuration for the firmware to read HAN data from that [...]

Since the documentation recommends using hardware serial, I assume UART[0-2] is preferred. I can see that UART0 maps to GPIO3, and UART1 maps to GPIO9, both which are unavailable on the LilyGo GPIO header. I was just curious if software serial is the only option when using ethernet?

ArnieO, 2024-02-20:
> Since the documentation recommends using hardware serial, I assume UART[0-2] is preferred. I can see that UART0 maps to GPIO3, and UART1 maps to GPIO9, both which are unavailable on the LilyGo GPIO header. I was just curious if software serial is the only option when using ethernet?

Newer ESP32 chips have a multiplexer, and can use HW serial on just about all GPIOs. So that will depend on your board.

gskjold, 2024-02-29:
T-Internet-POE uses GPIO17 (UART2-TX), so UART2 cannot be used with this board.

gskjold, 2024-02-29:
> THX, hmmmm still rebooting, but much slower...

Have you tried erase-flash before flashing new version?

gellfuen, 2024-03-02:
correct, did the erase before and tried that 2 times..

regards

[8 more replies]

closed 2024-03-21 by gskjold