UtilitechAS/amsreader-firmware #369: AmsToMqttBridge in Raw data (bytes) mode

roberthaugen73, 2022-12-02:
Hello.

Thank you for a good integration :-)

Swapped a Tibber Pulse(running local) with an AmsToMqttBridge in Raw data (bytes) mode.

Expected that the AmsToMqttBridge in Raw data (bytes) mode should present the data in the same format, but it differs.

IN MQTT Explorer:

Tibber Pulse data:
��+!�#���������7
����
�Kamstrup_V0001 �����
�57065672813xxxxx ���`���
�6841131BN2431xxxxx �������� ������� ������� �������� �������S ���3���K ���G��� ��� ���� ���4���� ���H����F

AmsToMqttBridge "Raw data (bytes) mode" data:
02190A0E4B616D73747275705F563030303109060101000005FF0A103537303635363732383133373631313509060101600101FF0A1236383431313331424E32343331303130343009060101010700FF06000000BD09060101020700FF060000000009060101030700FF060000000009060101040700FF0600000104090601011F0700FF060000004C09060101330700FF060000005109060101470700FF060000000909060101200700FF1200EE09060101340700FF1200EC09060101480700FF1200EC

The AmsToMqttBridge is timing out during configuration.

Meter manufacturer: Kamstrup
Meter type: 6841131BN243101040
OBIS List version identifier: Kamstrup_V0001

The AmsToMqttBridge is working in HomeAssistant mode.

Could not decode meter message: 30323139304130453442363136443733373437323735373035463536333033303330333130393036303130313030303030354646304131303335333733303336333533363337333233

roberthaugen73, 2022-12-02:
The messages are cut/paste from MQTT Explorer

gskjold, 2022-12-02:
Raw mode is as intended, dumping data in hex string to MQTT. I have never used Tibber pulse with MQTT, so it has not been my intention to copy them in any way. But I understand the confusion with the naming. Could rename it and add another option called binary. Not sure how far up my list that would be though, a bit busy with next version :)

roberthaugen73, 2022-12-02:
By Design. Not an error.

closed 2022-12-02 by roberthaugen73