UtilitechAS/amsreader-firmware #372: [bug] ENTSO-E menu point missing from all pages bug the homepage

LordMike, 2022-12-06:
**Describe the bug**
I've noticed that the "ENTSO-E API" menu is visible on the homepage, but is missing from other pages.

**To Reproduce**
Steps to reproduce the behavior:
1. Visit the AMS pages
2. Observe that the menu Configuration shows "ENTSO-E API"
3. Open any other page (Web, Ntp, ..)
4. Observe that the Configuration menu does not have "ENTSO-E API"

**Expected behavior**
I expected the menu to be present always. I was looking for it because in my first walkthrough I recalled seeing f.ex. Tokens for ENTSO-E, but couldn't find it.. Until I happened to click the menu on the homepage, at which point it was present.

**Screenshots**
[image]

**Hardware information:**
 - Country: DK
 - Meter: Kamstrup
 - Encryption enabled: yes
 - AMS reader: POW-K+
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - Version: 2.1.14
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: N/A
 - ENTSO-E API enabled: no

**Additional context**
I looked at the code and found that the menu is always present, but `d-none` is removed if the device has 32 KB or more. This codepath is only [hit if `fetch()` is called](https://github.com/gskjold/AmsToMqttBridge/blob/33bd3da31007afe18e3cdd7c3c7d4db5dcab60dd/web/application.js#L648-L650), which in turn only happens if [`setupChart()` is called](https://github.com/gskjold/AmsToMqttBridge/blob/33bd3da31007afe18e3cdd7c3c7d4db5dcab60dd/web/application.js#L425) - which happens [if there are any charts on the page](https://github.com/gskjold/AmsToMqttBridge/blob/33bd3da31007afe18e3cdd7c3c7d4db5dcab60dd/web/application.js#L201-L205) - which only appears on the frontpage.

ArnieO, 2022-12-07:
Thank you for reporting! 
But this is a known minor bug that will be fixed in v2.2, which is out for beta testing here: https://github.com/gskjold/AmsToMqttBridge/discussions/370
You are very welcome to test and comment!

gskjold, 2023-01-05:
Fixed in v2.2

closed 2023-01-05 by gskjold