UtilitechAS/amsreader-firmware #373: additional baud rate for smart meters with infrared interface

tqma1, 2022-12-07:
i want to use it with the Siemens TD-3511 smart meter which has an infrared interface which uses the following parameters:
300 baud, 7 bits, 1 stop bit and even parity
can you please implement this?

Best regards,
Thomas

gskjold, 2022-12-07:
Added in release that will be release any minute now. No guarantees it works with your setup though, not heard of any that use it with IR. But I see it uses IEC62056-21, so there is a possibility

closed 2022-12-07 by gskjold

ArnieO, 2022-12-07:
@tqma1 It will be interesting to know if the firmware works on an optical interface - please get back and give us an update!

gskjold, 2022-12-07:
http://www.mitterbaur.at/amis-leser.html

A quick skim through the source code, I see some serial.write, so it may need to request the data from the meter

ArnieO, 2022-12-07:
> So it will not work with my smart meter but great work from you!

That is right. This firmware will (currently) only work on meters configured for "Data push".