UtilitechAS/amsreader-firmware #374: [docs] Document the distribution network setting

LordMike, 2022-12-07:
**Is your feature request related to a problem? Please describe.**
See https://github.com/gskjold/AmsToMqttBridge/discussions/371#discussioncomment-4332212

**Describe the solution you'd like**
Some docs on the wiki about what to set (or where to find the info). I set mine to 230V, but given the discussion, I should probably have set 400V.. It had more consequences than I'd imagined.

Additionally, a link to the specific wiki page on the AMS webpage itself would be _very_ helpful :)

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

ArnieO, 2022-12-07:
Yeah... this is in fact a bit tricky. Most users have no idea whether they have a 230V or a 400V installation. In most cases the consequence of selecting the wrong one is only scaling of graphics, your case is a bit out of the ordinary - with a 3-phase meter installed in a 2-phase 400V installation. 

Some newer installations I have seen in Norway has a nice sticker in the fuse cabinet saying (for instance) "400V / TN", but that is far from being the case everywhere. 

If anyone reading this has a good idea on how to (easily) find out whether an installation is 230 V (IT/TT) or 400V (TN), we shall be most happy to include in the Wiki!

In the coming new firmware v2.2 (out for test [here](https://github.com/gskjold/AmsToMqttBridge/discussions/370)) there are links from sections in the config page to relevant sections in the Wiki. So it's coming! 😄

ArnieO, 2022-12-07:
This is what a Norwegian grid company says: https://www.foie.no/kundeservice/kundeforhold/volt

[image]

So: You need to check the small print on your power meter. 
If 400V is mentioned, it is a 400V installation. 
@gskjold : I propose to note this for a Wiki update?

gskjold, 2023-01-05:
Updated

closed 2023-01-05 by gskjold