UtilitechAS/amsreader-firmware #376: Price per hour out of the sheet

Lin4fun, 2022-12-10:
Hi,

this is less an issue, more a optical improvement. 
due to very high prices the price per hour goes out of the range of the sheet. Is it possible to make the ranges dynamic?
[image]

ArnieO, 2022-12-10:
Not sure what you mean here? 
The range is dynamic, and scales with the price.
Do you have an example showing the price graph goes out of range?

Lin4fun, 2022-12-12:
You see that on the picture provided. The range goes to 4.5 but the graph up to 5.63. So may it is just to adjust that the graph is inside the range.

ArnieO, 2022-12-12:
OK, understood: You wold prefer more horizontal lines. I think this is just the way this graphics library works when it auto scales! 😃 
Anyway, the graphics is about to change now, with v2.2 that is currently out for beta testing, see: https://github.com/gskjold/AmsToMqttBridge/discussions/370
So we will not do any adjustments on the graphics of v2.1.
You are welcome to test and comment on the beta version of v2.2!