UtilitechAS/amsreader-firmware #377: Discrepancy vs daily energy consumption data stored in Elhub

ArnieO, 2022-12-11:
**Describe the bug**
The values shown as "Energy use last month (kWh)" are in line with data recorded in Elhub. 
Elhub is the central data base in Norway where all collected meter data are stored. Individuals can log in on https://elhub.no/ (Min side) and download historic hourly consumption recordings.

Example: 
My consumption for 1st and 2nd December:
[image]

But AmsToMqttBridge reports 78 and 81 kWh:
[image]

- I have checked whether the error could be a one-hour shift in either direction: It is not.
- I have checked the hourly data last 24 hours - and they are identical to Elhub values.

ArnieO, 2022-12-12:
I have analysed a bit more, and believe the issue could be linked to a one-hour shift of data:

On the left are registrations imported from Elhub. 
On the right are values from our firmware (http://[IP adr]/dayplot.json)
[image]

Note that the graphics is correct, the last value 6.8 shown in the first hour of today. But in the dayplot.json the values are incorrect, as i23 should be the hour slot 23:00 to 00:00.
If the summing of a day is done based on the values in dayplot.json there is a one-hour shift. 

For the moment, I turn on the water heater at midnight - which could explain why my red curve above is too high. 

(There seems also to be some deviations between the data sets that are larger than rounding errors? But I think that is a minor issue.)

gskjold, 2022-12-12:
The data in dayplot.json is always in UTC time and is adjusted on client based on client timezone. 
The data points for monthplot is not based on the dayplot data, it keeps track of day to day accumulated value and calculates accordingly. Because of this, it is very important that the timezone is correct in Config -> NTP. Should be UTC+1 and 1hr summer offset for you.

ArnieO, 2022-12-12:
OK, thank you for explaining. 
So the source of the discrepancy must then be elsewhere... 🤔

gskjold, 2022-12-12:
I have not signed up for elhub before, how does this export work? The numbers seems useless to me....
[image]
[image]

gskjold, 2022-12-12:
By the way, I have compared AmsReader data with the data I have in OpenHAB, and those at least match

gskjold, 2022-12-12:
Seems to be a blend of kwh an wh... If I modify the sheet a bit, the sum matches yesterday:
[image]
[image]

I suggest you keep the kwh reading from 00:00 a couple of days in a row and calculate what the value should be for you. Could it be that elhub is wrong? I know that Kamstrup reports the timestamp differently from Aidon, although I cant remember the difference right now...

ArnieO, 2022-12-12:
> Could it be that elhub is wrong? I know that Kamstrup reports the timestamp differently from Aidon, although I cant remember the difference right now...

Eureka! I think this is what happens. (Have to dig more, but I really think this is it.)
**IF** so, it is a potential a major issue with the prices we have now, where many of us move load to lower cost hour slots.

**IF** so, it is not Elhub that is wrong, but the Kamstrup time stamps.

ArnieO, 2022-12-12:
> Could it be that elhub is wrong? I know that Kamstrup reports the timestamp differently from Aidon, although I cant remember the difference right now...

Interesting theory, but ... the customer that notified me uses a Pow-U, and he is with Lyse kraft. So probably an Aidon. And I see the same issue as he does, the numbers do not match.

AND see now that I have made a conversion errors with my CSV file, as I did not do the conversion as indicated here above. I incorrectly assumed the number of decimals in Elhub was constant. I now see that they record for instance 4,30 with only one decimal: "4,3". 

I will redo my CSV file conversions and get back.

gskjold, 2022-12-12:
I am not. Tried to log into the app to see if they would show me data anyway since they are my grid operator, but no...

I would make sure that this other guys timezone is set correctly