UtilitechAS/amsreader-firmware #380: Font support for smaller size (low resolution screens, or resized browser – graphs)

andreas1974, 2022-12-13:
**Describe the bug**
Adjacent numbers get on top of each other, which is very hard to read on a phone with small screen.
A attach a screen shot from a Sony Xperia X Compact (Android), where this problem is shown.

[image]

I notice a similar effect just resizing my Chrome web browser on Windows below a certain width. I attach two screen shots from my computer as well. One slightly problematic and one normal:

Too narrow:
[image]

Normal:
[image]

Perhaps every other number could be shown higher up and every other lower down? Or allow shrinking to an even smaller font size, on the graphs that are affected.

Bästa hälsningar
Andreas

PS Vi är så nöjda med detta gränssnitt och familjemedlemmarna tittar på elanvändningen och graferna med priser flera gånger varje dag! Tack för ett gediget arbete! DS

ArnieO, 2022-12-13:
Hej Andreas,
Tack for gode ord! 😃 

Thank you for your feedback and comments, but there will be no more adjustments on the graphical layout on firmware v2.1.x, as we are about to change the graphics library for v2.2 which is currently out for beta testing, please see: https://github.com/gskjold/AmsToMqttBridge/discussions/370 

It would be great if you could test the latest beta test version and see if it works better for you - and participate in the discussions on that thread.

gskjold, 2022-12-13:
Definitely issues on smaller screens with the beta versions as well, but it will be addressed

gskjold, 2023-05-17:
Graphs in v2.2 have improved in later patches which I consider to be good enough for mobile screens

closed 2023-05-17 by gskjold