UtilitechAS/amsreader-firmware #381: entso-e api - variabel "fee addition" based on hours ?

KlavsKlavsen, 2022-12-14:
**Is your feature request related to a problem? Please describe.**
I am always frustrated when the price shown in AMS graph is not the actual prices I pay - but only the "pool prices" - excl. the taxes etc. paid pr. kwh (in Denmark).
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
From january 2023 - new prices ordering is put in place - where depending on the hour of the day - a different "pricing" is added.
https://radiuselnet.dk/elnetkunder/tariffer-og-netabonnement/prisaendringer/

So from 00-06 the price is 21,27 (on top of pool price from entso-e ) (in winter - its diff. price in summer months)
from 07-17 and 20-23 - its 63,79 and from 17-20 - its 191,35

So it would be really nice instead of just a price multiplier - to have support for a table of "rules".. 
like if [month] (where month pref. could be a range) and [hour] (again a range) - then add X
Then I could add above rules - for both summer and winterperiods - and when calculating it would simply run through the ifs and "add those that match" ?

I don't think there will ever be need for anything but a month and an hour range (and both couuld be * - meaning you can have a set addition - and the addition could ofcourse be a "addition" or "multiplier" - so you can still support the current multiplier - where that applies.

gskjold, 2022-12-14:
Duplicate of #252

ArnieO, 2022-12-14:
I agree that the price presentation could be vastly improved - but as the firmware is used in many countries now, this could become very complex. Even inside one country (at least here in Norway) the grid tariff pricing models differ between the grid companies; and that is both the general model and the parameters within the pricing model. 

It is a significant challenge to develop a model that can take care of all that!
The best idea I have seen so far is some kind of "calculator" where the user defines parameters (A=123, B= 234, C=1.25 and so on) and then sets up an equation where the parameters are used, such as 
=[spotprice] * C + A + B
But still... there are those models that are time-of-day dependent; how do you model that in a calculator? 
=if ([now] between 06:00 and 12:00) then ... and so on. 

It could become a quite complex equation parser!

I would love to find a good and generic way to improve the pricing calculation - but I'm afraid it could become extremely complex, to the point where I'm not sure if it is worth the work.

KlavsKlavsen, 2022-12-14:
But I gave you a solution that I believe could easily be extended to fit all needs.

Simple table of rules - with 2x if's (optional) - on (to start with: either month-range or hour-range) - and simple 'add or multiply' rule - thats applied if the 1 or 2 if's match.
Its how many workflow engines work - and I see this as no different - only with 'price from entso-e api given as the thing you either 'add x to' or 'multiply with x' if rule(s) match.

You could then store rulesets - per provider/country if you like - to allow users to select pre=defined calculation rules - they can select if they do not want to define their own