UtilitechAS/amsreader-firmware #382: Several ”unknown” in import from Pow-P1

nossaile, 2022-12-14:
It looks like several entities are not imported from Pow-P1 and are marked “unknown” by HA. These entities are: 
•	Current month peak 2
•	Current month peak 3
•	L1 active export
•	L1 active import
•	L1 power factor
•	L2 active export
•	L2 active import
•	L2 power factor
•	L3 active export
•	L3 active import
•	L3 power factor

Below MQTT-explorer’s view of the payload:
AMSdata
prices = {"id":"58:CF:79:A4:36:36","prices":{"0":5.9310,"1":6.2447,"2":5.9321,"3":5.3946,"4":5.1547,"5":4.3420,"6":3.6849,"7":2.1758,"8":3.2508,"9":2.8345,"10":2.6611,"11":2.6447,"min":2.1758,"max":6.2447,"cheapest1hr":"2022-12-12T22:00:00Z","cheapest3hr":"2022-12-13T00:00:00Z","cheapest6hr":"2022-12-12T22:00:00Z"}}
energy = {"tPI":2404.95,"tPO":0.00,"tQI":1241.21,"tQO":0.09,"rtc":1670862990}
power = {"lv":"LGF5","id":"","type":"E360","P":6933,"Q":2358,"PO":0,"QO":0,"I1":15.20,"I2":6.70,"I3":12.00,"U1":228.80,"U2":229.40,"U3":227.30}
realtime = {"max":8.1,"peaks":[8.10],"threshold":10,"hour":{"use":5.26,"cost":31.21,"produced":0.00},"day":{"use":96.83,"cost":435.79,"produced":0.00},"month":{"use":518.91,"cost":2107.79,"produced":0.00}}
state = {"id":"58:CF:79:A4:36:36","name":"Pow-P1","up":156950,"vcc":3.302,"rssi":-75,"temp":-127.00,"version":"v2.1.15"}
I cannot find them in the payload.

All HW/SW with the latest firmware and software as of today.

gskjold, 2022-12-14:
Since you avoided the bug template, I can only assume that these values are not reported by your meter

nossaile, 2022-12-14:
The meter is a Landis&Gyr E360LTE supplied by Eon

ArnieO, 2022-12-15:
> Since you avoided the bug template, I can only assume that these values are not reported by your meter

It would be strange if L1, L2 and L3 active import are not reported from the meter...?

gskjold, 2022-12-15:
I have not seen many meters reporting active import/export per line. And L&G have different formats in different countries, so not sure about this one. But since this is P1 I am assuming DSMR, so I will also have to check if we extract those values from DSMR payloads, there could be an issue there.

nossaile, 2022-12-15:
Country=Sweden

nossaile, 2022-12-15:
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: ams-cf58 IP:zzzz Mac address:zzzz
* Free Heap RAM: 117288
* ESP SDK version: v4.4.1-1-gb8050b365e
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter

* Please type the command and press enter to execute.(? or h for this help)
***
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(V) (unwrapData)(C0) DSMR frame:
(V) 2F 4C 47 46  35 45 33 36  30 0D 0A 0D  0A 30 2D 30
(V) 3A 31 2E 30  2E 30 28 32  32 31 32 31  35 31 35 35
(V) 30 35 30 57  29 0D 0A 31  2D 30 3A 31  2E 38 2E 30
(V) 28 30 30 30  30 32 37 32  31 2E 31 32  36 2A 6B 57
(V) 68 29 0D 0A  31 2D 30 3A  32 2E 38 2E  30 28 30 30
(V) 30 30 30 30  30 30 2E 30  30 30 2A 6B  57 68 29 0D
(V) 0A 31 2D 30  3A 33 2E 38  2E 30 28 30  30 30 30 31
(V) 33 38 34 2E  33 33 36 2A  6B 56 41 72  68 29 0D 0A
(V) 31 2D 30 3A  34 2E 38 2E  30 28 30 30  30 30 30 30
(V) 30 30 2E 30  39 30 2A 6B  56 41 72 68  29 0D 0A 31
(V) 2D 30 3A 31  2E 37 2E 30  28 30 30 30  34 2E 31 39
(V) 30 2A 6B 57  29 0D 0A 31  2D 30 3A 32  2E 37 2E 30
(V) 28 30 30 30  30 2E 30 30  30 2A 6B 57  29 0D 0A 31
(V) 2D 30 3A 33  2E 37 2E 30  28 30 30 30  30 2E 32 39
(V) 32 2A 6B 56  41 72 29 0D  0A 31 2D 30  3A 34 2E 37
(V) 2E 30 28 30  30 30 30 2E  30 30 30 2A  6B 56 41 72
(V) 29 0D 0A 31  2D 30 3A 32  31 2E 37 2E  30 28 30 30
(V) 30 31 2E 39  38 30 2A 6B  57 29 0D 0A  31 2D 30 3A
(V) 32 32 2E 37  2E 30 28 30  30 30 30 2E  30 30 30 2A
(V) 6B 57 29 0D  0A 31 2D 30  3A 34 31 2E  37 2E 30 28
(V) 30 30 30 30  2E 33 35 34  2A 6B 57 29  0D 0A 31 2D
(V) 30 3A 34 32  2E 37 2E 30  28 30 30 30  30 2E 30 30
(V) 30 2A 6B 57  29 0D 0A 31  2D 30 3A 36  31 2E 37 2E
(V) 30 28 30 30  30 31 2E 38  35 35 2A 6B  57 29 0D 0A
(V) 31 2D 30 3A  36 32 2E 37  2E 30 28 30  30 30 30 2E
(V) 30 30 30 2A  6B 57 29 0D  0A 31 2D 30  3A 32 33 2E
(V) 37 2E 30 28  30 30 30 31  2E 30 37 32  2A 6B 56 41
(V) 72 29 0D 0A  31 2D 30 3A  32 34 2E 37  2E 30 28 30
(V) 30 30 30 2E  30 30 30 2A  6B 56 41 72  29 0D 0A 31
(V) 2D 30 3A 34  33 2E 37 2E  30 28 30 30  30 30 2E 32
(V) 36 35 2A 6B  56 41 72 29  0D 0A 31 2D  30 3A 34 34
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 56
(V) 41 72 29 0D  0A 31 2D 30  3A 36 33 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 30 30  2A 6B 56 41  72 29 0D 0A
(V) 31 2D 30 3A  36 34 2E 37  2E 30 28 30  30 30 31 2E
(V) 30 34 34 2A  6B 56 41 72  29 0D 0A 31  2D 30 3A 33
(V) 32 2E 37 2E  30 28 32 33  30 2E 37 2A  56 29 0D 0A
(V) 31 2D 30 3A  35 32 2E 37  2E 30 28 32  33 30 2E 31
(V) 2A 56 29 0D  0A 31 2D 30  3A 37 32 2E  37 2E 30 28
(V) 32 33 31 2E  30 2A 56 29  0D 0A 31 2D  30 3A 33 31
(V) 2E 37 2E 30  28 30 30 39  2E 37 2A 41  29 0D 0A 31
(V) 2D 30 3A 35  31 2E 37 2E  30 28 30 30  31 2E 39 2A
(V) 41 29 0D 0A  31 2D 30 3A  37 31 2E 37  2E 30 28 30
(V) 30 39 2E 32  2A 41 29 0D  0A 21 32 31  35 34 0D 0A
(V)
(V) (readHanPort)(C0) Received valid DSMR at 0
(V) (EnergyAccounting) Adding 0.0108 kWh import
(V) (EnergyAccounting)  and 0.0006 SEK
(V) (EnergyAccounting)  calculating threshold, currently at 1
(V) (EnergyAccounting)  new threshold 1
(D) (AmsWebServer)Serving /data.json over http...

gskjold, 2022-12-15:
Thanks, it does have these values, so must be the interpreter that does not extract them properly. I'll look into this

```
/LGF5E360

0-0:1.0.0(221215155050W)
1-0:1.8.0(00002721.126*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00001384.336*kVArh)
1-0:4.8.0(00000000.090*kVArh)
1-0:1.7.0(0004.190*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.292*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0001.980*kW)
1-0:22.7.0(0000.000*kW)
1-0:41.7.0(0000.354*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0001.855*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0001.072*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:43.7.0(0000.265*kVAr)
1-0:44.7.0(0000.000*kVAr)
1-0:63.7.0(0000.000*kVAr)
1-0:64.7.0(0001.044*kVAr)
1-0:32.7.0(230.7*V)
1-0:52.7.0(230.1*V)
1-0:72.7.0(231.0*V)
1-0:31.7.0(009.7*A)
1-0:51.7.0(001.9*A)
1-0:71.7.0(009.2*A)
!2154

```

closed 2022-12-19 by gskjold

nossaile, 2022-12-20:
The six for active power are correct now, but power factor values are wrong and the two for month peak are still marked "unknown". For the power factor values you could either:
1.  Remove the values
2.  Calculate correct values (voltage x current / active power) or
3.  Mark them "unknown"