UtilitechAS/amsreader-firmware #383: Multiple AMS readers

frafall, 2022-12-14:
**Is your feature request related to a problem? Please describe.**
I have two AMS readers, under Home Assistant they show up with identical descriptive text such as "Accumulated active Import" 
which makes these hard to separate as this text propegates into Energy UI. Currently I solve this by editing every sensor description.

**Describe the solution you'd like**
Would it be possible to be able to define a prefix in the amsleser mqtt configuration for use by all texts?

ArnieO, 2022-12-15:
Agreed; there should be an identifier in the HA MQTT payload that uniquely identifies each reader. 
Could adding the MAC-address in the payload be a good solution?

gskjold, 2022-12-15:
I am not that familiar with HA. The sensors are not grouped together per unit? Also, how is this solved by other types of devices?

frafall, 2022-12-16:
I have configured my two ams sensors to publish on separate topics "ams/oppe" and "ams/nede" to get unique values into home assistant. This gives me two distinct devices to use in the home assistant energy panel. BUT, the friendly name used in the panel is "Accumulated active Import" for both of them making it difficult to differentiate them. 

The actual devices I renamed fittingly "AMS/HAN 2b" and "AMS/HAN 2c" in home assistant so the sensors are named sensor.ams_han_2b_tpi and sensor.ams_han_2c_tpi. 

BUT, the friendly name showing up for the sensors are coming from amstomqttbridge, hardocded in the file src/mqtt/HomeAssistantStatic.h and therefore identical for both ams devices.

So, in my energy panel I get two "Accumulated active Import" without any easy means of differentiating which device/sensor they refer to.

I have solved this by adding a friendly name to all sensors with a prefix, ex "[2b] Accumulated active Import".

My suggestion would be to have an option to add a  prefix to use on all friendly names in the MQTT configuration panel with a default value of "" (empty). This would leave functionality unbroken. 

-morten

gskjold, 2023-04-01:
Included in next patch

closed 2023-04-01 by gskjold