UtilitechAS/amsreader-firmware #388: It seems that nordpool and entso has a small difference in prices (NO3) even with a mva factor of 1.25 in multiplier added

Ontrackx, 2022-12-17:
Nordpool 1.70NOK
ENTSO      1.94NOK

Checked my power delivery company vibb, and  vibb says 1,66
 at 10:00

gskjold, 2022-12-17:
Indeed, all system prices are given in EUR/MWh, what number you end up with in NOK depends on exchange rate you use. Would be good to be in sync, but I have no idea what strategy they use to choose exchange rate.

ArnieO, 2022-12-17:
But that difference is too much to be due to exchange rate?

tronde-ams, 2022-12-17:
The price published by Nordpool and Entso-e for NO3 is 129.7 EUR /MWh.
Did the interface fetch correct prices?

[image]
[image]

gskjold, 2022-12-19:
I agree the different sounds a bit much, I guess it probably is something else then.

ArnieO, 2022-12-19:
Our price data seems correct, but the exchange rate for conversion to local currency seems to be different compared to Nordpool. 
@Ontrackx : Could you elaborate on your findings? Your price difference seems not explainable by exchange rate differences. (You get a JSON with prices by atting /energyprices.json to your URL):
[image]

[image]

gskjold, 2023-01-05:
Closed due to inactivity

closed 2023-01-05 by gskjold