UtilitechAS/amsreader-firmware #389: FR: Support for IR port Kamstrup meter(s) with simple authentication

bipsendk, 2022-12-18:
**Is your feature request related to a problem? Please describe.**
Apparently no support for DLMS/COSEM meter with simple authentication via serial/IR Port 

**Describe the solution you'd like**
Support for meters with IR port (through serial port), where simple authentication is being used - e.g. Kamstrup 382Lx7

**Describe alternatives you've considered**
Using Gurux software on Raspberry Pi, but this will require quite an amount of programming to feed data into HA via MQTT

**Additional context**
Protocol description attached - IR port specs/auth/client/server addresses is listed at page 14
[55121424_D1_GB.pdf DLMS generation M N1.pdf](https://github.com/gskjold/AmsToMqttBridge/files/10254494/55121424_D1_GB.pdf.DLMS.generation.M.N1.pdf)

ArnieO, 2022-12-18:
Duplication, see this thread: https://github.com/gskjold/AmsToMqttBridge/issues/126
Requires Kamstrup-proprietary protocol, only seen partially documented on various personal implementations.

gskjold, 2022-12-19:
> Duplication, see this thread: #126

Are we sure this is the same? When I look at the document document, this is standard IEC62056-7-5, but with two-way communication.

> 55121424_D1_GB.pdf DLMS generation M N1.pdf

gskjold, 2022-12-19:
I see now that this document is also referenced in the other thread. I think I may have seen this document before, but it's hard to keep track these days.

closed 2022-12-19 by gskjold

ArnieO, 2022-12-19:
> Are we sure this is the same? When I look at the document document, this is standard IEC62056-7-5, but with two-way communication.

Sorry - my mistake. I meant this thread: https://github.com/gskjold/AmsToMqttBridge/issues/373

My opinion is that reading "IR eye" / optical port is outside the scope of this firmware, as I consider those meters to be obsolete. They will surely be replaced by AMS meters at some point of time.