UtilitechAS/amsreader-firmware #390: No support for Wifi SSID with no password

bwentzel, 2022-12-23:
**Describe the bug**
Currently the PSK password is required

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'wifi settings'
2. remove PSK password and clear
3. Click save
4. Not able to save without password entered

**Expected behavior**
Save without a password to connect to wifi network without password

**Relevant firmware information:**
 - Version: 2.1.16

**Additional context**
This is required for a project where we need to connect to a public network that has separate security mechanisms that still protects the device.

ArnieO, 2022-12-25:
Thank you - nice catch! You're probably the first to have tried on an SSID without password.  
I believe @gskjold is currently on vacation - but he'll surely look into this when he's back.

gskjold, 2023-01-05:
Will be possible in v2.2

closed 2023-01-05 by gskjold