UtilitechAS/amsreader-firmware #392: No login

skjeldgaard, 2022-12-26:
I have updated the security settings, with username and password.
When accessing the device you are not prompted for at login, neither is a url for login provided in documentation, leaving the device UI impossible to access.

ArnieO, 2022-12-27:
Which firmware version is this? 
Could it be your browser that blocks popups?

skjeldgaard, 2022-12-27:
Running Edge on WorkPC - can't get it to work.... Works in Chrome.
Maybe leave note in documentation :)
thx!

closed 2022-12-27 by skjeldgaard

skjeldgaard, 2022-12-27:
> Did you find out what the issue was with Edge? Some browser setting that
> blocked? We would need to know in order to make a note in the doc - thanks!

Not really sure, not able to get it to open the login popup - might be to
do with some settings being controlled by my organization.

best regardds

ArnieO, 2022-12-27:
Ah, ok - that makes sense.