UtilitechAS/amsreader-firmware #394: 1 hour offset between HA and device

joachimtingvold, 2022-12-29:
**Describe your problem**
Energy usage is offset by 1 hour when comparing HA with AmsToMqttBridge website. 22-23 on device becomes 23-00 in HA, 23-00 on device becomes 00-01 in HA, etc. All clocks seems to be the same, and all timezone settings seems to be correct as well.

I assume the "Summertime offset = +1hr" setting is automatic (i.e. when set to "+1hr" it will add/remove 1 hour depending on wether or not its summertime, without having to manually do something).

Maybe I'm missing something really obvious here? Or?

[image]
[image]
[image]
[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Encryption enabled: no
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - Version: 2.1.16
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: N/A
 - ENTSO-E API enabled: N/A

ArnieO, 2022-12-29:
Interesting. 
At first glance this looks like an HA integration issue, but we have quite many HA users so it is surprising if this has not been spotted by others. (I don't use HA, so I cannot check/verify)

**Other HA users: Do you see the issue, or have an idea what could be the problem?**

Yes, summertime is automatic and works the way you describe.

Maybe related:
I have had a similar issue on my RasPi that runs Octopi/OctoPrint, where the GUI showed correct time but the calculation of end of print was 1 hour off. Please see if you can find a helpful hint here: https://community.octoprint.org/t/wrong-time-in-the-browsers-tab-title/29373/2

kjetilrong, 2023-01-01:
Same issue here.  "AMS reader v2.1.16",  Payload:Home-Assistant,  og Home Assistant 2022.12.8

[image]
[image]

Could it have something to do with MQTT timestamp?

bardahlm, 2023-01-01:
This is an issue with HA. I belive it is caused by HA assigning the measurements to the hour it receives the measurment. And as the HAN port only outputs kWh once an hour (at 10 or so seconds past the hour) everything is offset in HA.

I've posted about this in the community forum but got no response: https://community.home-assistant.io/t/energy-one-hour-off/490506

The best fix would be if NVE revised their policy and got kWh meter readings posted more often.

ArnieO, 2023-01-02:
> This is an issue with HA. I belive it is caused by HA assigning the measurements to the hour it receives the measurment. And as the HAN port only outputs kWh once an hour (at 10 or so seconds past the hour) everything is offset in HA.

That makes sense. But are you sure it is a HA reception time issue - or could it be fixed by adjusting the timestamp coming from AmsToMqttBridge? If so, I guess it could be done in the HA MQTT payload generation for the cases where there is 1 hour (or more) since last payload. 

> The best fix would be if NVE revised their policy and got kWh meter readings posted more often.

Agree, but would imply changing firmware inn all AMS meters in Norway; I don't think it is realistic - and could create problem for other AMS meter readers running firmware that is less adaptive than this one.

tronde-ams, 2023-01-02:
Can it be possible to do some trickery with the realtime accumulated kWh instead for Home Assistant?

bardahlm, 2023-01-02:
> That makes sense. But are you sure it is a HA reception time issue - or could it be fixed by adjusting the timestamp coming from AmsToMqttBridge? If so, I guess it could be done in the HA MQTT [...]

I'm quite sure this is a HA issue. I'll try opening an issue for this.

bardahlm, 2023-01-18:
> I'll try opening an issue for this.

Can't really call this a success! I'll guess we'll have to come up with a PR to get this fixed and that is above my Python skills.

kng, 2023-01-19:
I looked at my meter (GMT+1) and it is accumulating correctly afaict, the past hour ~3.5kWh is correctly going to the graph.
Ticking over from 8:59-9:00 (not all the top row fields updated...)

[image]

[image]

But the graph in HA doesn't show the last value but shifts it forward ?!

[image]

joachimtingvold, 2023-01-19:
> I looked at my meter (GMT+1) and it is accumulating correctly afaict, the past hour ~3.5kWh is correctly going to the graph.
> […]
> But the graph in HA doesn't show the last value but shifts it forward ?!

Yes, that is literally why this issue was created (shows correctly in AmsToMqttBridge, but 1 hour offset/delayed in HA).

closed 2023-03-11 by gskjold