UtilitechAS/amsreader-firmware #396: Option to Turn off LED

jca1981, 2023-01-03:
Would be nice if we could turn off the blue blinking LED., my bedroom lights up every 10 secunds at night :)

bjornsivertsen, 2023-01-03:
You got your electric meter in your bedroom?!

ArnieO, 2023-01-03:
I understand your frustration! Since such placement of power meter you describe is most common in Denmark, I assume you are Danish, and have a Pow-K on a Kamstrup meter.

In Sweden, meters are often placed in an outdoor enclosure with access from outdoor. In Norway the meters are in an indoor steel cabinet together with the fuses. You are less fortunate - and I am sorry for keeping you awake!

On Pow-U and Pow-K, the blue part of the LED is directly connected to the incoming data signal, it is not controlled by the microcontroller. The intention is to have a clear indication on whether there is incoming data from the meter, without involving the controller. This means that this issue can only be fixed in hardware - not in firmware.

I have had other comments on this, and understand this is a real problem for some of you. It might be remedied in future redesign by adding components. I'm afraid your best option is the very "low-tech" solution to put some black tape over the small surface mounted LED:
[image]

jca1981, 2023-01-03:
i will do that, thanks :)

henriklund, 2023-01-07:
As another Dane, all the places I have lived / visited, the predominant placements have been the utility room, an outdoor enclosure or in the hallway (in immediate connection to the front door). So far I am still to experience a meter in the bedroom. I think it would drive me nuts to have the meters diode flash all the time. @jca1981 you have my sympathy.

jca1981, 2023-01-08:
Thanks @henriklund, i will try to put some electrical tape on the LED. would be nice for future revision to have a switch or something to turn the light off. would it maybe be possible to cut the contact point with a razerblade or unsolder the LED?

jca1981, 2023-01-08:
> You got your electric meter in your bedroom?!

small arpartment in copenhagen, powermeter is right in front of the bedroom, in a small hallway.

ArnieO, 2023-01-08:
> Thanks @henriklund, i will try to put some electrical tape on the LED. would be nice for future revision to have a switch or something to turn the light off. would it maybe be possible to cut the contact point with a razerblade or unsolder the LED?

Not sure why I did not think of this earlier - but if you remove R3 (1k resistor), the blue LED will be gone without any consequence for the function of the board.
[image]

svennek, 2023-03-18:
Just to confirm, I found this when I was on my way in here to ask to remove "disco mode" from the amsreader. I too live in a small (Copenhagen) apartment, and while it isn't in my bedroom, I can still see the disco effect in all of my apartment at night. The power thru the leds must be really impressive as this devices lights up my appartment more than 20+ other devices combined... 

I am going to remove R3 as suggested and remove the gpio config for the green led. Luckily removing the led is no biggie for me.

For turning off the green led, a great option would be, that the software controlled the green light and used it for say 300 seconds after boot and then turned it off. After debugging bootup the green led usually does not have any use-case.

ArnieO, 2023-03-19:
The green LED is only used during reboot, so I think your suggested firmware modification will have no effect.

The problem for some users is the blue LED, and the solution for that is mentioned here above. Users that do not have a soldering iron could consider taking a more drastic approach and use a pair of cutter pliers and crush the resistor. Just make sure to remove the debris. This approach might of course be non-reversible.

svennek, 2023-03-19:
@ArnieO That does not match how my Power-K+ behaves, mine is blue for a few seconds and then a short (but sharp) green and then darkness (off) for 7ish seconds, so the green is definitely not just used just for boot for mine. It has been online for weeks and still blinks green in the end of the blue.

My power-k+ was delivered beginning of this year and due to a lot of "fun" with the local utility first installed in February.

ArnieO, 2023-03-19:
Ah... sorry - you're of course right, sorry about that! There is a green confirmation blink every time it recognizes a valid payload.

@gskjold : I think it is a good proposition to implement this as proposed.
Maybe implemented as a new option in the **User interface** tile: "Green LED" with the options "ON" and "OFF after 300 sec".

svennek, 2023-03-19:
Just FYI, did the mod where I desoldered R3 and as expected the blue light stopped and the device itself seems otherwise fully functional.

ArnieO, 2023-03-19:
Yes, that works just fine. You could desolder also the resistor for the green part of the LED, but you'll then loose the visual confirmation of accepted payload and also the yellow blinking when device is in AP mode (it will blink red instead).  So better to wait for a firmware update.

ArnieO, 2023-03-20:
This article points a finger at this very problem of too bright status LEDs. 

So as the designer I can just bend my head and say: **_Mea culpa! Mea very much culpa...!_**

https://hackaday.com/2020/02/20/we-ruined-status-leds-heres-why-that-needs-to-change/

Quote:
_A status LED, most would agree, is there to indicate status. It need only deliver enough light to be seen when observed by a querying eye. **What it need not do is glow with the intensity of a dying star, or illuminate an entire room for that matter.**_

OK, ok... I get it! 😅

closed 2023-12-25 by gskjold