UtilitechAS/amsreader-firmware #401: Energy price per hour unit

kng, 2023-01-17:
The price reported is in EUR (or whatever is set in the ENTSO-E) and should be in unit/kWh instead ?
[image]

This is set in HA Energy dashboard:
[image]

ArnieO, 2023-01-17:
I'm not sure I understand your comment. 
If you go to Config/Entso-E you can choose currency. Did you select EUR there?
[image]

kng, 2023-01-17:
It's not whether it is SEK or EUR, but SEK/kWh or EUR/kWh.

kng, 2023-01-17:
To clarify, the price is fetched in currency/MWh as seen [here](https://github.com/gskjold/AmsToMqttBridge/blob/master/src/entsoe/EntsoeApi.cpp#L64-L65), then converted to currency/kWh. This unit should be carried over to the unit presented over [ha](https://github.com/gskjold/AmsToMqttBridge/blob/master/src/mqtt/HomeAssistantStatic.h#L45)/json etc

gskjold, 2023-01-23:
Any chance you could test if this is ok in latest build in #370 ?

kng, 2023-01-23:
Yes, it's up and running 2.2 beta and after switching the currency and rebooting it is no longer complaining.
[image]

[image]

closed 2023-01-23 by kng