UtilitechAS/amsreader-firmware #402: request: Use DHCP hostname for HA discovery documents

LordMike, 2023-01-17:
**Is your feature request related to a problem? Please describe.**
I just updated to 2.1.16, and disabled mDNS, because I'm running DHCP and a full DNS server. Additionally, I'd like my devices to end up in `.c.home`, rather than `.home`. 

Now there is an _ever so slight inconvenience_... :) 

The HA discover doc still reports the configuration url as `http://ams-cf58.local/`, where I think it will eventually stop working. It should probably have been `http://ams-cf58.c.home/`. :)

**Describe the solution you'd like**
Adjust the HA configuration url based on the actual hostname+domain.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

LordMike, 2023-02-14:
Just chiming in to say it's still relevant in `2.2.8` :)

gskjold, 2023-02-16:
I am hesitating to do this kind of change for a patch release, so I have scheduled it for a future minor. I have yet to decide the release date of that minor

gskjold, 2023-04-01:
Implemented possibility to configure HA hostname

closed 2023-04-01 by gskjold