UtilitechAS/amsreader-firmware #404: Missing bootloader filer for ESP32-S2

svenderikv, 2023-01-19:
Hi
I have used AmsToMqttBridge on a ESP32 and it worked perfectly. Now I want to use a ESP32-S2 board but cant find the bootloader file. The wiki is mentioning bootloader_qio_40m.bin but I am unable to find it.
The zip-file for the ESP32 have the bootloader_dio_40m.bin but no such file in the ESP32-S2 zip-file.
Thanks for a very good piece of software :-)

svenderikv, 2023-01-19:
Actually, I was aware of it, bu the link doesn't work.
[image]
There is a bootloader_qio_40m.elf but is this the same as the bin-file?

gskjold, 2023-01-19:
Not sure what the deal is with that repo, the files are sometimes not there. But if you change to tag 2.0.3 (which is the version we currently run) you will find them https://github.com/espressif/arduino-esp32/tree/2.0.3/tools/sdk/esp32s2/bin

svenderikv, 2023-01-19:
Thanks a lot :-) I will try to flash it right away :-)

ArnieO, 2023-01-19:
> Not sure what the deal is with that repo, the files are sometimes not there. But if you change to tag 2.0.3 (which is the version we currently run) you will find them https://github.com/espressif/arduino-esp32/tree/2.0.3/tools/sdk/esp32s2/bin

@gskjold If the Espressif source is unstable, maybe store copies of the bootloader files on the AmsToMqttBridge repo?

gskjold, 2023-01-23:
These will be correctly included in the zip file starting from v2.2. Will need to update documentation at that point.

closed 2023-02-03 by svenderikv