UtilitechAS/amsreader-firmware #413: Time and date off / ntp not working?

Lin4fun, 2023-01-28:
**Describe the bug**
The time and date is off, looks like there is something wrong with ntp import. This was also the case in version 2.1.7, now upgraded to 2.2.0. I have now issues with other devices in the network using ntp service from no.pool.ntp.org. 
Time sone is set to Europe/oslo, NTP server no.pool.ntp.org. Also tried setting IP of ntp pool aswell a local running ntp service and through dhcp.
Shown date is 28.00.23 time started at 18.00

**Screenshots**
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Encryption enabled: no
 - AMS reader: Pow-U+

**Relevant firmware information:**
 - Version: 2.1.7 & 2.2.0
 - MQTT: yes
 - MQTT payload type: [e.g. JSON]
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: yes

ArnieO, 2023-01-29:
Unknown problem, the default setting has always worked well on my router - and we should expect to have heard other users reporting this.
Could it be a setting in your router or firewall that gives you these problems?

gskjold, 2023-01-29:
I've been running no.pool.ntp.org and the "obtain from DHCP" option unchecked for the last day now, seems to work fine here... Not sure what else to test to be honest, but I'll circle back to this if I can think of anything

atian888, 2023-01-31:
I am new to this AMSreader. Running vs 2.2.2 and get timelabel as "01.00.1970 01:13". 
Testet no.pool.ntp.org as well without success.

ArnieO, 2023-02-01:
This is my setting - that works for me:
[image]

atian888, 2023-02-01:
Strange, but today the date/time was normal (!)... Maybe some delay somewhere... 
Chip: esp32s2
Device: Pow-U+

Lin4fun, 2023-02-02:
Seems to work fine now after upgrade to 2.2.5

closed 2023-02-02 by Lin4fun