UtilitechAS/amsreader-firmware #414: Webpage broken after update to 2.2.0

stone17, 2023-01-28:
**Describe the bug**
Webpage only shows top header (Status of HAN, ESP and so on), but no graphs or anything else.
It is possible to manually navigate to /configuration.
MQTT still sends all info about power consumption.

The console in the browser shows this error:
VM91:1 Uncaught (in promise) SyntaxError: Bad escaped character in JSON at position 361

Seen this behavior on 2 devices.

**To Reproduce**
Upgrade from 2.1.17 to 2.2.0 on ESP32-S2-MINI-1

**Expected behavior**
Webinterface keeps working as on 2.1.17

**Screenshots**
[image]

**Hardware information:**
 - Country: Sweden
 - Meter: Dont know
 - Encryption enabled: Dont know
 - AMS reader: ESP32-S2-MINI-1
 - M-bus adapter (if applicable): NA

**Relevant firmware information:**
 - Version: 2.2.0
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: GPIO18?
 - HAN baud and parity: 115200 8N1 inverted
 - Temperature sensors: NA
 - ENTSO-E API enabled: no

gskjold, 2023-01-28:
Sounds like a invalid character in data.json, could you get me the content of that?

stone17, 2023-01-28:
how do i fetch the json?

gskjold, 2023-01-28:
Go to the network tab in the developer tools and you can click on the file and go to the response tab

stone17, 2023-01-28:
{"im":20000,"om":0,"mf":16,"i":546,"e":0,"ri":0,"re":160,"ic":1934.300,"ec":0.004,"ric":8.929,"rec":524.765,"u1":225.00,"u2":226.80,"u3":228.90,"i1":1.00,"i2":1.20,"i3":0.50,"f":0.00,"f1":0.00,"f2":0.00,"f3":0.00,"v":0.000,"r":-72,"t":-127.00,"u":9089,"m":114040,"em":1,"hm":1,"wm":1,"mm":1,"me":0,"p":null,"mt":255,"ds":0,"ea":{"x":1.8,"p":[1.52,2.07,2.01,1.74,1.50],"t":5,"h":{"u":0.02,"c":0.00,"p":0.00,"i":0.00},"d":{"u":14.71,"c":0.00,"p":0.00,"i":0.00},"m":{"u":352.64,"c":0.00,"p":0.00,"i":0.00}},"pr":"","he":0,"ee":0,"c":1674932510,"a":true}

gskjold, 2023-01-28:
Hum weird, looks totally find... What if you click on the error message in the console, does it expand to the problem in question?

stone17, 2023-01-28:
[image]

msasdk, 2023-01-28:
I have the exact same issue.

stone17, 2023-01-28:
In this view it is highlighting a backslash when i click on the error.

gskjold, 2023-01-28:
So the slash in the model number breaks sysinfo.json, got it... That's a bummer... Too bad this didn't show up in testing. I will make some adjustments

gskjold, 2023-01-28:
Here is a test build. Unplug the device from the meter and hit reset so that the model number is cleared. Unzip the attached file and then go to the info page and upload the firmware.bin file

This is for ESP32-S2, did both of you have the same chipset?
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/10528092/esp32s2.zip)

gskjold, 2023-01-28:
Hang on, I made a mistake in the code...

gskjold, 2023-01-28:
Here we go: 
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/10528098/esp32s2.zip)

stone17, 2023-01-28:
After flashing your zip file, it works again. It does want to update to 2.2.0 though.

gskjold, 2023-01-28:
Yeah, all beta firmware will show the upgrade link to the latest version. I will publish this update soon

stone17, 2023-01-28:
No worries, thanks a lot for the quick support :)

closed 2023-01-28 by gskjold

msasdk, 2023-01-28:
I'm on regular ESP32 and 2.2.1 still does not work unfortunately.

gskjold, 2023-01-28:
Navigate to /sysinfo.json and /data.json on your device and attach the content in a new issue

msasdk, 2023-01-28:
sysinfo
{"version":"v2.2.1","chip":"esp32","chipId":"c4ef","mac":"24:0A:C4:05:EF:C4","apmac":"24:0A:C4:05:EF:C5","board":203,"vndcfg":true,"usrcfg":true,"fwconsent":1,"hostname":"ams","booting":false,"upgrading":false,"net":{"ip":"192.168.1.49","mask":"255.255.255.0","gw":"192.168.1.1","dns1":"192.168.1.1","dns2":"0.0.0.0"},"meter":{"mfg":255,"model":"\\253868807_B","id":""},"ui":{"i":0,"e":0,"v":0,"a":0,"r":0,"c":0,"t":0,"p":0,"d":0,"m":0,"s":0},"security":0}

data
{"im":11085,"om":0,"mf":16,"i":1237,"e":0,"ri":0,"re":249,"ic":9125.513,"ec":0.001,"ric":2.346,"rec":1827.630,"u1":229.40,"u2":230.20,"u3":234.10,"i1":0.40,"i2":4.10,"i3":1.20,"f":0.00,"f1":0.00,"f2":0.00,"f3":0.00,"v":0.000,"r":-63,"t":-127.00,"u":1337,"m":224372,"em":1,"hm":1,"wm":1,"mm":1,"me":0,"p":null,"mt":255,"ds":2,"ea":{"x":5.0,"p":[1.51,7.33,1.65,7.30,7.28],"t":10,"h":{"u":0.21,"c":0.00,"p":0.00,"i":0.00},"d":{"u":9.78,"c":0.00,"p":0.00,"i":0.00},"m":{"u":421.18,"c":0.00,"p":0.00,"i":0.00}},"pr":"10Y1001A1001A47J","he":0,"ee":0,"c":1674938872,"a":true}

gskjold, 2023-01-28:
Sort of looks like the same issue, but shouldnt happen... if you are in chrome, press F12 to open developer tools, check console to see if there are errors like the screenshot earlier in this thread

msasdk, 2023-01-28:
[image]

msasdk, 2023-01-28:
This is what 2.2.0 looked like (firefox)
[image]

gskjold, 2023-01-28:
Are you able to go to configuration? Check if all gui elements are set as hidden. If so, you need to enable some of them. Normal config:

[image]

msasdk, 2023-01-28:
Yep sorry about that :) 
All good now.

gskjold, 2023-01-28:
Sweet! Good to hear