UtilitechAS/amsreader-firmware #415: Monthly-cost error after upgrade from 2.1.6 to 2.2.1

atle-dahlman, 2023-01-29:
The monthly cost error after the update.  Illustrated below with an image from Home assistant.

[image]

gskjold, 2023-01-29:
Sorry it disappeared on you. v2.2 have gone through extensive testing before release and this has never happened. Might be a edge case around loading and saving.

atle-dahlman, 2023-01-29:
The funny thing is that it just didn't get reset.  If so the daily and monthly cost should be the same.  It's not.
[image]

gskjold, 2023-01-29:
Found the issue, it incorrectly divided the stored value for month by 100..

closed 2023-01-29 by gskjold