UtilitechAS/amsreader-firmware #417: Circular graphs

mnorrelund, 2023-01-30:
Thanx for your work. Her some graphical hickups from chrome & firefox browser - no zoom level:
[image]
[image]
[image]
last one is as expected. Red/green? I have solarpannels attached

gskjold, 2023-01-30:
Thank you! Fix will be included in next patch

ArnieO, 2023-01-30:
> last one is as expected.

Not quite. The graph should be scaled according to your installed maximum production power. Please check that you have provided correct value for your installation here:
[image]

gskjold, 2023-01-30:
Scaling is broken in current version...

ArnieO, 2023-01-30:
> Scaling is broken in current version...

💡

closed 2023-01-31 by gskjold