UtilitechAS/amsreader-firmware #418: Pricing for DK2 reported in GUI or through MQTT does not match Nordpool

henriklund, 2023-01-31:
**Describe the bug**
Pricing reported in GUI or through MQTT does not match Nordpool

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Configuration'
2. Set Price Region to DK2, currency to DKK and multiplier to 1
3. Enable 'Price fetch' and save configuration

**Expected behavior**
Pricing to be alike Nordpool.

**Screenshots**
Nordpook shows:
[image]

However, pricing shown is less than DKK 0,01. Through MQTT (GUI shows a similar picture) following data can be seen:
`Message 14 received on home/sensors/power/meter at 8:03:
{
    "id": ".......",
    "prices": {
        "0": 0.0085,
        "1": 0.0083,
        "2": 0.008,
        "3": 0.0075,
        "4": 0.0072,
        "5": 0.0071,
        "6": 0.0071,
        "7": 0.0071,
        "8": 0.0071,
        "9": 0.0074,
        "10": 0.0073,
        "11": 0.007,
        "min": 0.0051,
        "max": 0.0085,
        "cheapest1hr": "2023-01-31T22:00:00Z",
        "cheapest3hr": "2023-01-31T20:00:00Z",
        "cheapest6hr": "2023-01-31T17:00:00Z"
    }
}`

**Hardware information:**
 - Country: DK
 - Meter: Kampstrup
 - Encryption enabled: yes
 - AMS reader: Pow-K ESP8266
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - Version: 2.2.3
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: GPIO3 (according to configfile)
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: N/A
 - ENTSO-E API enabled: yes (although ESP8266 does not provide possibility to enter ENTSO-E API key)

**Additional context**
N/A

henriklund, 2023-01-31:
Just an observation: seems that if I set multiplier to 100, output meets expectation. I do not know if that is due to source data being buggy or something in the code.

gskjold, 2023-01-31:
Turns out it was a server side problem. If you reboot now, it should fetch price correctly

henriklund, 2023-01-31:
I can confirm reboot resolved the issue. Thanks!

closed 2023-01-31 by henriklund