UtilitechAS/amsreader-firmware #420: MQTT reporting system status

henriklund, 2023-01-31:
**Describe your problem**
I use a standard MQTT JSON for receiving data. On the previous version I had installed (v2.1.14) this worked as a charm. However, after v.2.2.3 I noted my graphs have started to get 'holes' in them. Further investigation shows that the reason is that the MQTT reports a normal data package (list 4) immediately followed by 'System data'. Since I pull the tPI information for the overall state of the entity, I had to create a workaround to get the value from previous message. This is not pretty since it relies on the assumption of a specific order in messages. Secondly I pull all data and apply those as the entities attribute. Those data (attributes) are used in various Grafana graphs. The introduction of a (almost) blank 'System data' message means that once every minute, attributes will be blank since the 'System data' message does not contain any relevant data (similar to list 4).

I thus wonder what purpose the 'System data' serves, and if there is any way to avoid this message? I do recognize that there is an option to use 'Home Assistant' format for MQTT. However, I created my setup prior to this, and it has been working flawlessly ever since (well until now). 

**Hardware information:**
- Country: DK
- Meter: Kampstrup
- Encryption enabled: yes
- AMS reader: Pow-K ESP8266
- M-bus adapter (if applicable): N/A
- Relevant firmware information:

**Version: 2.2.3**
- MQTT: yes
- MQTT payload type: JSON
- HAN GPIO: GPIO3 (according to configfile)
- HAN baud and parity: 2400 8N1
- Temperature sensors: N/A
- ENTSO-E API enabled: yes (although ESP8266 does not provide possibility to enter ENTSO-E API key)

henriklund, 2023-01-31:
A quick idea; maybe an option (when JSON is selected) allowing to chose if message should be split into different topics ({Publish topic}/messagename; eg. home/sensors/power/meter/systemdata, home/sensors/power/meter/list1). That could either be per message type or an 'all or nothing'.

henriklund, 2023-02-01:
A quick'n'dirty hack to workaround this behavior (just remember to change [[[topic]]] to the correct topic) :
- create an automation to split MQTT messages into separate topics:

```
alias: mqtt splitter
description: ""
trigger:
  - platform: mqtt
    topic: [[[topic]]]
condition:
  - condition: template
    value_template: "{{ trigger.payload_json.version is not defined }}"
action:
  - service: mqtt.publish
    data:
      topic: [[[topic]]]/data
      payload: "{{ trigger.payload }}"
mode: single
```

gskjold, 2023-02-01:
Having separate topics for JSON sounds like a good idea. I'll keep that in mind for a future patch

closed 2024-03-25 by gskjold