UtilitechAS/amsreader-firmware #421: Power usage off by a lot, starting at a midnight.

RealElwood, 2023-02-01:
At midnight the reader started reporting kW numbers that don't add upp with the U*I numbers, see picture.
I've since discovery updated to latest fw, rebooted, 

[image]

[image]

**Hardware information:**
 - Country: Sweden
 - Meter: Kamstrup Omnipower
 - Encryption enabled no
 - AMS reader: esp8266

**Relevant firmware information:**
 - Version: 2.2.4
 - MQTT: yes
 - MQTT JSON
 - HAN baud and parity: 2400 8N1
  - ENTSO-E API enabled: no

ArnieO, 2023-02-01:
There are some meters in Sweden that report an extra digit on power, leading to a 10x error.
Please check the value on the LCD display of your meter and compare - to see if this is the case for you.
We have not found a way to identify this issue in the code - so if this is the case you need to correct it by setting the power multiplier to 0.1. 
[image]
You should also check if Energy (kWh) has a 10x scale error. 

Maybe a firmware update was pushed to your meter at midnight?
Anyone else in Sweden seeing the same thing?

PS: With firmware 2.2 you can now get prices also on your ESP8266 based device! 
https://twitter.com/amsleser/status/1620373179977060354?s=20

gskjold, 2023-02-01:
If you had v2.1 series firmware before, you are one of the few Kamstrup users affected by a change in the code. Ref release notes for 2.2.0: https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.2.0

gskjold, 2023-02-01:
Hmm, that's strange. I'll look into how that might happen.

gskjold, 2023-02-01:
Do you remember what version you upgraded from?

RealElwood, 2023-02-01:
> Do you remember what version you upgraded from?
2.1.7 I think, but that was after it started reporting wrong. The multiplier was just there today...