UtilitechAS/amsreader-firmware #422: Upgrade to 2.2.4  loses meter GPIO settings

cmock, 2023-02-01:
**Describe the bug**
Upgraded from 2.1.7 to 2.2.4, afterwards the GPIO pin configuration for the meter interface was lost (reset to UART0 in the GUI), was UART2/GPIO16 before.

**To Reproduce**
Steps to reproduce the behavior:
1. run 2.1.7 happily
2. Upgrade to 2.2.4 via the web interface
3. Watch a red "HAN" indicator

**Hardware information:**
 - Country: Austria
 - Meter: SagemCom T210-D
 - Encryption enabled yes
 - AMS reader: ESP32
 - M-bus adapter (if applicable): resistive divider

**Relevant firmware information:**
 - Version: 2.2.4
 - MQTT: yes
 - MQTT payload type: raw (full)
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8E1]
 - Temperature sensors none
 - ENTSO-E API enabled: no

The problem is reproducible. I downgraded to 2.1.7, lost all data in the process, reconfigured everything and tried again, same behaviour.

Config file (secrets removed) from 2.1.7 is attached.
[configfile.txt](https://github.com/UtilitechAS/amsreader-firmware/files/10556782/configfile.txt)

ArnieO, 2023-02-01:
There seems to be some issue when upgrading on your hardware setup. 

While waiting for a fix I would try the following:

- Try to upload that config from v2.1.17 to firmware v2.2.4
- If that does not work: Reflash firmware over USB as explained in the Firmware v2.2 User Manual chapter 9:
https://amsleser.no/index.php?controller=attachment&id_attachment=45

This will give you a "virgin" installation, where you at the "Initial configuration screen" (paragraph 9.2, point 13) can select "Generic hardware", then GPIO16.
[image]
[image]

Then save and reboot. 

You will now find a new section on the configuration page, where you can adjust GPIO settings:
[image]

gskjold, 2023-02-01:
Found the issue. It should actually affect all hardware upgrading from v2.1.x, but if you reboot straight after the upgrade, the cleared GPIO config is not persisted.

closed 2023-02-01 by gskjold