UtilitechAS/amsreader-firmware #426: build error

frahlg, 2023-02-04:
**Describe the bug**

When I am trying to compile I get the following error. Regardless of which computer, I have tried both on my Mac M1 and a Windows 11 PC and the error is the same, it seems that there is a path missing. 

```

FileNotFoundError: [WinError 3] The system cannot find the path specified: 'lib/SvelteUi/app/dist':
  File "C:\Users\fredr\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 195:
    env.SConscript(item, exports="env")
  File "C:\Users\fredr\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\fredr\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\fredr\Downloads\amsreader-firmware-2.2.5\amsreader-firmware-2.2.5\lib\SvelteUi\scripts\generate_includes.py", line 48:
    for filename in os.listdir(webroot):
```

frahlg, 2023-02-04:
I can add to the comment above that I have no problem compiling version 2.1.14.

ElMyggo, 2023-02-05:
I get the same errors when i try to build 2.2.5.... worked in old version 2.1.17 for me

GuvHas, 2023-02-05:
I too get the same error when compiling.

gskjold, 2023-02-05:
Commited some files, try again

GuvHas, 2023-02-05:
> Commited some files, try again

Now it works for me, thanks for the quick solution!

ElMyggo, 2023-02-05:
Works here too :-)

frahlg, 2023-02-05:
Thanks, seems to work for me. Now just need to figure out how to compile for my esp32-c3. :)

closed 2023-02-05 by frahlg