UtilitechAS/amsreader-firmware #427: esp32-c3-mini build job

frahlg, 2023-02-04:
I have bought a p1-reader from https://smart-stuff.nl/ , which is based upon the RISC-V, ESP32-C3-MINI-1 board. It runs just fine with esphome but I would like to to have the functionality of amsreader (I already have an esp8266 running). At the moment I have problems compiling the FW, and it would be greatly appreciated if a binary for the esp32-c3-mini could be added in the release build section.

frahlg, 2023-02-05:
I have now tried to compile the FW for the esp32-c3-mini-1 board using the following settings (I had better luck with a newer version of the tasmota platform below). And it seems to be compiling until `Compiling .pio/build/esp32-c3/libbea/AmsDecoder/GcmParser.cpp.o`.

## platformio.ini
`[env:esp32-c3]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.01.01/platform-espressif32.zip
framework = arduino
board = esp32-c3-devkitm-1
board_build.mcu = esp32c3
build_flags = ${common.build_flags} -DFRAMEWORK_ARDUINO_C3
lib_ldf_mode = off
lib_compat_mode = off
lib_deps = ${esp32.lib_deps}
lib_ignore = ${common.lib_ignore}
extra_scripts = ${common.extra_scripts}`

## Build errors

`Compiling .pio/build/esp32-c3/libbea/AmsDecoder/GcmParser.cpp.o
lib/HwTools/src/HwTools.cpp: In member function 'void HwTools::getAdcChannel(uint8_t, AdcConfig&)':
lib/HwTools/src/HwTools.cpp:122:18: error: 'ADC1_CHANNEL_5_GPIO_NUM' was not declared in this scope
             case ADC1_CHANNEL_5_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:122:18: note: suggested alternative: 'ADC1_CHANNEL_2_GPIO_NUM'
             case ADC1_CHANNEL_5_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
                  ADC1_CHANNEL_2_GPIO_NUM
lib/HwTools/src/HwTools.cpp:124:34: error: 'ADC1_CHANNEL_5' was not declared in this scope
                 config.channel = ADC1_CHANNEL_5;
                                  ^~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:124:34: note: suggested alternative: 'ADC1_CHANNEL_4'
                 config.channel = ADC1_CHANNEL_5;
                                  ^~~~~~~~~~~~~~
                                  ADC1_CHANNEL_4
lib/HwTools/src/HwTools.cpp:126:18: error: 'ADC1_CHANNEL_6_GPIO_NUM' was not declared in this scope
             case ADC1_CHANNEL_6_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:126:18: note: suggested alternative: 'ADC1_CHANNEL_2_GPIO_NUM'
             case ADC1_CHANNEL_6_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
                  ADC1_CHANNEL_2_GPIO_NUM
lib/HwTools/src/HwTools.cpp:128:34: error: 'ADC1_CHANNEL_6' was not declared in this scope
                 config.channel = ADC1_CHANNEL_6;
                                  ^~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:128:34: note: suggested alternative: 'ADC1_CHANNEL_4'
                 config.channel = ADC1_CHANNEL_6;
                                  ^~~~~~~~~~~~~~
                                  ADC1_CHANNEL_4
lib/HwTools/src/HwTools.cpp:130:18: error: 'ADC1_CHANNEL_7_GPIO_NUM' was not declared in this scope
             case ADC1_CHANNEL_7_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
Archiving .pio/build/esp32-c3/lib998/libAmsDataStorage.a
lib/HwTools/src/HwTools.cpp:130:18: note: suggested alternative: 'ADC1_CHANNEL_2_GPIO_NUM'
             case ADC1_CHANNEL_7_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
                  ADC1_CHANNEL_2_GPIO_NUM
lib/HwTools/src/HwTools.cpp:132:34: error: 'ADC1_CHANNEL_7' was not declared in this scope
                 config.channel = ADC1_CHANNEL_7;
                                  ^~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:132:34: note: suggested alternative: 'ADC1_CHANNEL_4'
                 config.channel = ADC1_CHANNEL_7;
                                  ^~~~~~~~~~~~~~
                                  ADC1_CHANNEL_4
lib/HwTools/src/HwTools.cpp:148:18: error: 'ADC2_CHANNEL_1_GPIO_NUM' was not declared in this scope
             case ADC2_CHANNEL_1_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:148:18: note: suggested alternative: 'ADC1_CHANNEL_1_GPIO_NUM'
             case ADC2_CHANNEL_1_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
                  ADC1_CHANNEL_1_GPIO_NUM
lib/HwTools/src/HwTools.cpp:150:34: error: 'ADC2_CHANNEL_1' was not declared in this scope
[... 96 lines cut ...]
             case ADC2_CHANNEL_8_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
                  ADC2_CHANNEL_0_GPIO_NUM
lib/HwTools/src/HwTools.cpp:178:34: error: 'ADC2_CHANNEL_8' was not declared in this scope
                 config.channel = ADC2_CHANNEL_8;
                                  ^~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:178:34: note: suggested alternative: 'ADC2_CHANNEL_0'
                 config.channel = ADC2_CHANNEL_8;
                                  ^~~~~~~~~~~~~~
                                  ADC2_CHANNEL_0
lib/HwTools/src/HwTools.cpp:180:18: error: 'ADC2_CHANNEL_9_GPIO_NUM' was not declared in this scope
             case ADC2_CHANNEL_9_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:180:18: note: suggested alternative: 'ADC2_CHANNEL_0_GPIO_NUM'
             case ADC2_CHANNEL_9_GPIO_NUM:
                  ^~~~~~~~~~~~~~~~~~~~~~~
                  ADC2_CHANNEL_0_GPIO_NUM
lib/HwTools/src/HwTools.cpp:182:34: error: 'ADC2_CHANNEL_9' was not declared in this scope
                 config.channel = ADC2_CHANNEL_9;
                                  ^~~~~~~~~~~~~~
lib/HwTools/src/HwTools.cpp:182:34: note: suggested alternative: 'ADC2_CHANNEL_0'
                 config.channel = ADC2_CHANNEL_9;
                                  ^~~~~~~~~~~~~~
                                  ADC2_CHANNEL_0
*** [.pio/build/esp32-c3/libfd3/HwTools/HwTools.cpp.o] Error 1`

gskjold, 2023-02-06:
Added esp32c3 profile in platformio.ini and necessary modifications. Let me know if it works for your board

gskjold, 2023-02-06:
Btw, the build flag FRAMEWORK_ARDUINO_C3, is that something you found from documentation? I cannot find it anywhere. I can only see a build flag for solo1 at https://github.com/tasmota/platform-espressif32/#stable-release-supporting-arduino-and-idf-443

frahlg, 2023-02-06:
It seems to build now, thanks! Need to wait until I get home until I can verify it actually works on the device, I will let you know. The build flag was just a creative idea as the solo1 had one and I tried a lot of different options. It seems to make no difference, works even without. So I guess you could remove this.

closed 2023-02-06 by frahlg

gskjold, 2023-02-10:
Did this build actually work?